aceinnolab/Inkycal #118: No '=' in line 's gonna have to be later than that. I&#39\'

emilyboda, 2020-05-29:
**Describe the bug**
When I run inkycal.py, I get the following:
```
pi@raspberrypi:~ $ python3 /home/$USER/Inky-Calendar/modules/inkycal.py
Cleanup of previous images...Done
Initialising weather... Done
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  No '=' in line 's gonna have to be later than that. I&#39\'
RSS module: Connectivity check passed. Generating image...Done
Current Date: 28 May 2020
Current Time: 23:50
-----------Main programm started now----------
Calibration.. not required. Continuing...
Weather module: Connectivity check passed, Generating image...Done
Calendar module: Generating image...Failed!
Error in Calendar module!
Reason:  No '=' in line 's gonna have to be later than that. I&#39\'
RSS module: Connectivity check passed. Generating image...Done
Initialising E-Paper...Done
Sending image data and refreshing display...Done
Sending E-Paper to deep sleep...Done
10 Minutes left until next refresh
Cleanup of previous images...Done
```

The screen shows the weather module and the rss feed module, but instead of the calendar module it shows the same words as the error above: 
`No '=' in line 's gonna have to be later than that. I&#39\'`

**To Reproduce**
I believe I've followed the installation steps to a tee and haven't made any modifications. I've only added my weather API key, 

This is the contents of my settings.py:

```
ical_urls = ["url1.ics", "url2.ics"] #(it has actual .ics urls)
rss_feeds = ["http://feeds.bbci.co.uk/news/world/rss.xml#"] # Use any RSS feed
update_interval = "60"                # "10" # "15" # "20" # "30" # "60"
api_key = "fdsajkflds;jafskl;"      # Your openweathermap API-KEY -> "api-key"
location = "Philadelphia, US"            # "City name, Country code"
week_starts_on = "Sunday"             # 'Sunday' # Monday
calibration_hours = [0,12,18]         # Do not change unlesss you know what you are doing
model = "epd_7_in_5"        # Choose the E-Paper model (see below)
language = "en"                       # "en" # "de" # "fr" # "jp" etc.
units = "imperial"                      # "metric" # "imperial"
hours = "24"                          # "24" # "12"
top_section = "inkycal_weather"       # "inkycal_weather"
middle_section = "inkycal_calendar"   # "inkycal_agenda" #"inkycal_calendar"
bottom_section = "inkycal_rss"        # "inkycal_rss"
```

I thought that maybe the issue had to do with [this issue](https://github.com/aceisace/Inky-Calendar/issues/115), and I did actually have some non-.ics urls in the ical list, but I removed them and rebooted the pi and I still have the same issues.

[here's a link to what it looks like)](https://photos.app.goo.gl/g8B6oh7zF8npH9cV9)

emilyboda, 2020-05-29:
I removed all my calendar links and replaced it with the provided one and there is no more errors. How can I test my personal calendar links to figure out why they are breaking the program?

`["https://calendar.google.com/calendar/ical/en.usa%23holiday%40group.v.calendar.google.com/public/basic.ics'"]`

aceisace, 2020-05-29:
@emilyboda 
Hi, nice to hear from you again :). The current release does not unfotunately show the full traceback on the epaper display or the console, which makes debugging a bit harder than expected. 

However, if the problem no longer occurs when removing non-default iCalendars, it means it's not related to inkycal_calendar, but rather the ics.py library (parsing of events). Ics.py is _too strict_ with validating iCalendars.

For the next release however (reaching BETA in the next few days), inkycal now uses the icalendar library, which so far has proven to be pretty reliable and forgiving with iCalendars.

emilyboda, 2020-05-29:
So it appears that even calendars that pass the iCalendar Validator website are causing my program to fail. I will keep trying to pinpoint the issue and see if any of my calendars work.

emilyboda, 2020-05-29:
Alright, so it appears that all of my calendars work except my main calendar. The validator website says "maximum file limit exceeded". I've been using google calendar since 2010, and that main calendar has the most events out of any. So I'm not sure what to do about that.

aceisace, 2020-05-29:
@emilyboda If you don't need the events from the previous years, you could technically remove them from Google Calendar (maybe all which are older than 3 years). That way, both the validator and the iCalendar parsing library should be able to parse the iCalendar correctly.

emilyboda, 2020-05-30:
@aceisace thanks for the idea! I think I'm going to go ahead and create a secondary "main" calendar for the time being, since I value my old information very highly and don't want to get rid of it. 

I am hoping that the package you are using for the calendars in your next BETA won't have the same problem parsing very large .ics files, do you think so?

aceisace, 2020-05-31:
@emilyboda That sounds like a better idea :).
The BETA does indeed use a different, more mature lib (iCalendar). It should technically be able to parse pretty large iCalendars, so it might be worth a try.