aceinnolab/Inkycal #122: AttributeError: 'Layout' object has no attribute 'supports_colour' when using black and white screen

emilyboda, 2020-06-05:
I receive an [AttributeError](https://photos.app.goo.gl/GUqZWuwyBpwtvwVf6)  when I have set my screen to:
`"model": "epd_7_in_5"`

The error no longer appears when I change the model to:
`"model": "epd_7_in_5_colour"`

I have fixed the error by adding an else statement to the following in layout.py:
```
      if 'colour' in model:
        self.supports_colour = True
      else:
        self.supports_colour = False
```

emilyboda, 2020-06-05:
This is meant for the BETA. Can I put problems I find with the BETA here or should I only put them in the discord?

aceisace, 2020-06-05:
@emilyboda Thanks for the report. I'll add a fix for this soon on the BETA branch. 
And yes, you can open up issues for a specific branch. I'll just add the [BETA] tag there to keep this separate. Discord is also fine :)

aceisace, 2020-06-07:
OK, pushed the fix for this issue in the BETA branch.
Closing this PR as this issue has been solved.
If you need to re-open this issue, please feel free to do so.

closed 2020-06-07 by aceisace