aceinnolab/Inkycal #126: Module Request: Todoist

RadCLarite, 2020-06-12:
Is anyone working on a module to link to Todoist or equivalent?  I have been looking at the various e-ink calendar systems and I think that is the critical capability. Thanks!

aceisace, 2020-06-12:
@RadCLarite Hi and thanks for your interest in the inkycal software :).
About the integration of the todoist, this is a feature that is in planning for the next release. I'm estimating the new release will be available in 1-2 month.

If you are interested in contributing, you can test the __[BETA](https://github.com/aceisace/Inky-Calendar/tree/dev_ver2_0)__ release already. Any help (e.g. testing) for this feature would allow a faster implementation and a more bug-free module. You can also join the [Discord server](https://discord.gg/4YWdkVT) to give feedback, suggestions, help with testing etc.

RadCLarite, 2020-06-12:
Looking forward to it! I just successfully installed the current release; I
am planning a second device, so I can probably beta test there.

aceisace, 2020-06-17:
@RadCLarite 
OK, I'm drafting the todoist_api module now. For now, this is the current state:
```python3
#!/usr/bin/python3
# -*- coding: utf-8 -*-

"""
todoist module for Inky-Calendar Project
Copyright by aceisace
"""

try:
  import todoist
except ImportError:
  print('todoist is not installed! Please install with:')
  print('pip3 install todoist-python')

api = todoist.TodoistAPI('your_token')
api.sync()

# Print name of author
print(api.state['user']['full_name']+'\n')

tasks = (task.data for task in api.state['items'])

for _ in tasks:
  # filter tasks which haven't yet been completed:
  if _['checked'] == 0:
    print('[ ] {}'.format(_['content']))

```
Could you test the module and give some feedback? For example, which tasks and how should they be displayed on the display?

RadCLarite, 2020-06-18:
First - awesome! The code runs fine with my token. It lists all the undone todos, but it doesn't list the dates or when it is due. Having those two things (date and time) would be nice. My opinion is that one should be able to select the "Projects" from which you want to list todos. For example, you don't need your grocery shopping list on a relatively non-portable calendar.

aceisace, 2020-06-18:
Thanks for your feedback @RadCLarite . So tasks should be categorized by the project name and sorted by due date. Only undone tasks should be displayed as the space on the display is limited.

You mentioned dates and time, are these two different for a task?

RadCLarite, 2020-06-18:
If there is a setting to select which projects are actually shown, that would be a nice feature. Given the limited space, you may not want your grocery list overtaking your display. 

due date and time - not all tasks have a due time, some are date only. I don't know how the data flows, but if there is no due time, then I would probably not attempt to display the associated time?
for example:
20 Jun: Order Father's day gift     <no due time>
21 Jun - 13:00: Pickup up Gift       <due date and time>

aceisace, 2020-06-19:
@RadCLarite 
Had some time to work on this. Below are the properties of a task:
```python3
added_by_uid
assigned_by_uid
checked
child_order
collapsed
content
date_added
date_completed
day_order
due
has_more_notes
id
in_history
is_deleted
labels
parent_id
priority
project_id
responsible_uid
section_id
sync_id
user_id
```

The ones related to date/time are:
```
date_added
date_completed
due
```
Is only the due date of any relevance from these properties?

So on the display, it should look something like:
{Project name}
{due date (and time)} {task name}
{due date (and time)} {task2 name}

RadCLarite, 2020-06-19:
Maybe the priority? I don't use it, but others may be interested.

yes to the display format, I think that would work well, maybe ranked by
priority? (I don't use the priority myself)

{Project name}
{due date (and time)} {task name}
{due date (and time)} {task2 name}

or:
{Project name}
{due date (and time)} {priority} {task name}
{due date (and time)} {priority} {task2 name}

aceisace, 2020-06-19:
OK, added filtering by groups, sorting by groups and due time. Could you test the following code?
```python3
#!/usr/bin/python3
# -*- coding: utf-8 -*-

"""
todoist module for Inky-Calendar Project
Copyright by aceisace
"""

try:
  import todoist
except ImportError:
  print('todoist is not installed! Please install with:')
  print('pip3 install todoist-python')

api = todoist.TodoistAPI('your_token')
api.sync()

# Print name of author
print(api.state['user']['full_name']+'\n')

# define a list of project names which should be displayed
project_filter = ['Inky-Calendar']

# Get all projects by name and id
all_projects = {project['name']: project['id']
                for project in api.projects.all()}

# Check if project from filter could be found
if project_filter:
  for project in project_filter:
    if project not in all_projects:
      print('Could not find a project named {}'.format(project))
      project_filter.remove(project)

# function for extracting project names from tasks
get_project_name = lambda task: (api.projects.get_data(
                                 task['project_id'])['project']['name'])

# If the filter is empty, parse all tasks which are not yet done
if project_filter:
  tasks = (task.data for task in api.state['items'] if (task['checked'] == 0)
           and (get_project_name(task) in project_filter))

# If filter is not empty, parse undone tasks in only those projects
else:
  tasks = (task.data for task in api.state['items'] if (task['checked'] == 0))

# Simplify the tasks for faster processing
simplified = [{'name':task['content'], 'due':task['due'],
               'priority':task['priority'], 'project_id':task['project_id']}
              for task in tasks]

# Group tasks by project name
grouped = {}

if project_filter:
  for project in project_filter:
    project_id = all_projects[project]
    grouped[ project ] = [
      task for task in simplified if task['project_id'] == project_id]
else:
  for project in all_projects:
    project_id = all_projects[project]
    grouped[ project ] = [
      task for task in simplified if task['project_id'] == project_id]

# Print tasks sorted by groups
for project, tasks in grouped.items():
  print('*', project)
  for task in tasks:
    print('{} {}'.format(
      task['due'] if task['due'] != None else '',
      task['name']))
```

To use it, replace the token and the list project_filter. Any string of a project name will be checked for existence, then parsed.

Also, how should the tasks be displayed? One task per line (like the RSS-module) or a table-like layout?

RadCLarite, 2020-06-29:
sorry - missed this notification, will test soon. You deserve 2 coffees.

RadCLarite, 2020-06-29:
I get an error when I replace the name of the project within the quotes -
for example 'Inbox' generates this output: (Sorry, ssh in, don't know how
to copy and paste from windows SSH)

[image: image.png]

However, when I replace it with blank '', it first says
"Could not find a project named"
then lists all projects, like this:
* Inbox
* Personal
....
If the project has a to do, it lists
* Project Name
{'date': '2020-08-04', 'is_recurring': false, 'lang': 'en', ' string': '
Aug 4','timezone':None}  actual task here

and repeats until the end of tasks.
So, it probably has something to do with how the project name is passed?

aceisace, 2020-06-30:
@RadCLarite Thank you very much for your coffee 😄 💯 
I'll fix the mentioned issues in the next coming days. Look like the string could be formatted better and the project name handling could be improved too :).

aceisace, 2020-07-05:
@RadCLarite OK, I've improved the code a tiny bit, it now lists something like this:
```bash
* Inky-Calendar # project name
•It works! # simple todo
•This is awesome! # another simple todo
•Todo with due 29 Juni 18:45 # date + timed todo
•Todo with date only 22 Juni #date todo (no timing)
```

I could not reproduce the Error with `Could not find a project named` when the list is empty like this: `[]`, but I've fixed the formatting of the date to be more easier to read.

I'll go ahead with making this an official inkycal-module now since the first draft is complete. Once I make some more progress, I'll let you know :)

aceisace, 2020-11-13:
@RadCLarite 
Finally got some time to work on Inykcal again. Apologies for the slow development. I've finsihed the first draft of the module for the next release, see [this commit](https://github.com/aceisace/Inky-Calendar/commit/8190325cb78effefe11e9701666bee8ab6b30773). A few things can be tweaked a bit more, but for now, this is a image generated by the new module:
[image]
I'll be closing this issue since this feature has now been implemented. If you feel a need to re-open this issue, please feel free to do so again. Thanks in advance.

closed 2020-11-13 by aceisace