aceinnolab/Inkycal #129: Bug: Settings.json Online Generator parses iCal URLs into an incorrectly formatted list and causes an error

emilyboda, 2020-06-29:
When you add two or more URLs to the iCal URL bar in the online settings.json file generator in the format requested, the file that is generated adds single quote between the URLs and double quotes at the beginning and end. 

For example, if you input the following into the online UI:
`'ulr_a.com','url_b.com','url_c.com'`

The file generated will show the following:
```
"ical_urls": [
			"ulr_a.com','url_b.com','url_c.com"
		 ]
```
Notice the single quotes between the URLs, but double quotes on the outer edges. This is an incorrectly formatted list and will result in an error if you try to use the iCal URLs in any module.

aceisace, 2020-06-30:
@emilyboda OK, this should hopefully be fixed now. The fix allows:
* Adding urls without quotes: `url1, url2, url3`
* Directly converts urls into a array (list) instead of string replacment
* Generated correct settings file, i.e.:
```python
                   "ical_urls": [
					"url1",
					"url2",
					"url3"
				]
```

aceisace, 2020-07-04:
OK, did some testing with the updated web-ui. The problem seems to be fixed now :). 
I'll be closing this issue. If you feel a need to re-open this issue, please feel free to do so.

PR #130 (2020-07-04): Fix web-UI url handling
Added option to select 9.7" epaper display (still work in progress)
Adding urls for icalendars and rss is now easier: (previously: 'URL1', 'URL2', 'URL3') -> now: comma separated urls: url1,url2, url3
Fixed an issue where adding multiple urls would lead to unexpected issues.

fixes #129
closed 2020-07-04 by aceisace