aceinnolab/Inkycal #136: Feature request: M ore calendar events

centauri61032, 2020-08-17:
Again, I'm new to this app. So forgive me if I'm all wet.

But am I correct in thinking that the calendar section is only showing events for the current calendar month? If so, would it be possible to show events for something like 30 days out (if space is available of course), instead of limiting to the current month? And speaking of space, could there be an option to allow the extra future calendar events to use the bottom section instead of having any rss feed? 

Thanks again!!

centauri61032, 2020-08-23:
OK, I am trying to test the BETA version, but having a problem...
I have been successful up to and including the point where I created a json file, and sent it to the pi. However, I don't understand what to do under the 'Running Inkycal' section. Kind of a noob here, so I don't understand how to do the 'run python' type things. Could you please spell out the steps in this section in a bit more detail?

Thanks again!!

aceisace, 2020-08-24:
@centauri61032 
With the flexibility of the BETA release, an out-of-the box running script was no longer required. To make the running script that runs Inkycal, create a new .py (python) file with the following content:

```python
# Open Python3 and import package
from inkycal import Inkycal

# If you see 'numpy is not installed, please install with pip3 install numpy', 
# please run the following command in the Terminal, then restart python and try again:
# pip3 uninstall numpy

# tell the Inkycal class where your settings file is
inky = Inkycal('/path/to/your/settings/file', render = True)
# render means rendering (showing) on the ePaper. Setting render = False will not show anything on the ePaper

# test if Inkycal can be run correctly, running this will show a bit of info for each module
inky.test()

# If there were no issues, you can run Inkycal nonstop:
inky.run()
```

After pasting the content, save the .py file (e.g. `inkycal_run.py`). Then you can start inkycal with: `python3 inkycal_run.py` and it should work.

centauri61032, 2020-08-24:
DUH. OK, got it.
I did what you said, updated the setting path, ran the script, and got this

====================

pi@raspberrypi:~/Inkycal $ python3 inkycal_run.py
Settings file OK!
Traceback (most recent call last):
  File "inkycal_run.py", line 9, in <module>
    inky = Inkycal('/home/pi/Inkycal', render = True)
  File "/home/pi/Inkycal/inkycal/main.py", line 42, in __init__
    self.Settings = Settings(settings_path)
  File "/home/pi/Inkycal/inkycal/config/settings_parser.py", line 87, in __init__
    self._settings['panels'] if _['location'] == 'top']
  File "/home/pi/Inkycal/inkycal/config/settings_parser.py", line 87, in <listcomp>
    self._settings['panels'] if _['location'] == 'top']
  File "/home/pi/Inkycal/inkycal/config/settings_parser.py", line 84, in <lambda>
    to_decimal = lambda percentage: percentage/100
TypeError: unsupported operand type(s) for /: 'NoneType' and 'int'
pi@raspberrypi:~/Inkycal $

=======================

Really sorry to be a noob. But what did I do wrong?
Thanks!