aceinnolab/Inkycal #137: Configuration: Weather location

centauri61032, 2020-08-18:
In the setup file, it says to define the weather location using the City and the Country Code.  How do you include the State (in the US)? As an example, there are multiple 'Monroe' cities in 'US'. What do you enter in the config to select Monroe, LA over Monroe, WI?

Thanks again!

centauri61032, 2020-08-23:
OK, thanks for the info. But unfortunately that doesn't solve the weather problem...

The problem is that there are duplicate CITY, COUNTRY combinations in the USA.  Just for an example, go to the openweathermap site you linked to, and search for 'Freeport, US'. Multiple locations will show up. And if you select any of they, they will all say 'Freeport, US' in the following screen. I need to know how to select THE Freeport, US at 42.297, -89.621. (Again, just using this as one of the many examples that we would have here in the US.)

If I do select the Freeport at those coordinates, the URL changes. It adds 'city/4893171' to the end of the URL. I am assuming that the number is some kind of city code that openweathermap uses to resolve this very issue on their site. Perhaps Inky-Calendar could also use this location ID code somehow?

If you have any ideas, let me know. Thanks!

aceisace, 2020-08-24:
@centauri61032 
Thanks for the info. Pyowm does have an option to use the city ID of openweathermap. There wasn't a record of where this could be useful in Inkycal, but now it makes sense to choose between the ID or the `City, Country` string. I'll change the topic of this issue and will implement this option soon.

aceisace, 2020-11-12:
@centauri61032 I've implemented support for location ID in [this commit](https://github.com/aceisace/Inky-Calendar/commit/ea5118517367efd8e301f4c1781069e5cc06fa3b). Some testing has been done to test it's functionality and so far it seems to be going well. This feature will be available for everyone in the next release.
Since this issue could be resolved, I'm closing it now. If you feel a need to re-open this issue, please feel free to do so again.

closed 2020-11-12 by aceisace