aceinnolab/Inkycal #140: Make panel colors and dimensions configurable (with a defaults based on the model)

Atrejoe, 2020-10-09:
A global settings for the panel color would be nice, especially for panels with different color versions (like the `epd_7_in_5_v3_colour` panel, which exists in BWY (yellow) and BWR (red) versions.

In the future more panels with more colors are imaginable.

Initial palettes could be derived from the type in most cases.

The order of the palette could also be meaningful for some implementations, like

- Background color (I guess will always white or effectively transparent)
- Foreground color (mainly black)
- Support color (yellow or red so far)

Panel dimensions could also be derived but overriden. (For instance to fit a frame)

For inspriation see:
https://github.com/Atrejoe/Inky-Calendar-Server/blob/develop/InkyCal.Utils/DisplayModelHelper.cs

See also the panel picker at:
https://inkycal.robertsirre.nl/

and panel config:
[image]

https://github.com/aceisace/Inky-Calendar/blob/a92bf492e0051b704dda42a32fb1184f227c5035/inkycal/modules/inkycal_image.py#L50

Atrejoe, 2020-10-11:
Very similar logic is also present within Inkycal, see:

https://github.com/aceisace/Inky-Calendar/blob/dev_ver2_0/inkycal/config/layout.py#L25

Also generalizing the support color to just 'color.

aceisace, 2021-02-12:
I've converted this issue into a card on the project page for better clarity of what needs to be implemented for the next release. You can track this feature here: [Colour palette](https://github.com/aceisace/Inkycal/projects/3#card-54823757)

Since this has been accepted as a feature request, I'll be closing this issue. If you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-12 by aceisace