aceinnolab/Inkycal #141: Calendar rendering is broken

vitasam, 2020-10-26:
The calendar rendering is broken:
```
..:~/Github/Inky-Calendar$ python3 runinkycal.py 
Settings file OK!
Weather loaded
Calendar loaded
RSS loaded
loaded inkycal
no validation implemented
no validation implemented
Generating images for all modules...
Error!
Traceback (most recent call last):
  File "/home/.../Github/Inky-Calendar/inkycal/main.py", line 180, in run
    exec(generate_im)
  File "<string>", line 1, in <module>
  File "/home/.../Github/Inky-Calendar/inkycal/modules/inkycal_calendar.py", line 261, in generate_image
    write(im_colour, event_lines[cursor], (date_width, line_height),
IndexError: list index out of range

OK

inkycal has been running without any errors for 0 display updates
Programm started just now
34 Minutes left until next refresh
```

The Calendar URL has been verified with iCalendar [validator](https://icalendar.org):
```
File Size: 814 lines, 24761 bytes
Number of events found: 34
Number of timezones found: 1
```

aceisace, 2020-11-22:
@vitasam I've found the issue in the calendar module causing this error and have fixed it in [this commit](https://github.com/aceisace/Inky-Calendar/commit/948573cb59b7bb86c8075f1a2d721ea49f171942). Some testing has been done too and it has so far proven to have fixed this specific issue.

Thanks you for your report!
I'm closing this issue now since the bugfix has been deployed. If you feel a need to re-open this issue, please feel free to do so again.

closed 2020-11-22 by aceisace

vitasam, 2020-11-22:
Thank you! I will test it later tonight

commit 948573cb (2020-11-22): Fixed #141 list index out of range

If the number of events exceeded the number of lines allocated for events, a *list index out of range* exception was thrown. This has been fixed now  by providing an additional check.