aceinnolab/Inkycal #149: Empty screen.

hiteshjoshi, 2020-12-14:
**Describe the bug**
My EPD 7.5 V2 from Waveshare works fine with the default waveshare demo's etc. But with Inkycal the screen is always blank.

Here is my terminal output.

```
pi@raspberrypi:~ $ python3 /home/$USER/inky_run.py
inkycal_image loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
generating image(s) for Inkycal Image - show an image from a URL or local path...OK!
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
Date: 14 Dec 20 | Time: 17:05
Generating images for all modules...successful
```

aceisace, 2020-12-24:
@hiteshjoshi 
I think I got back to you on Discord. Can you post an update of the current progress? 
Thanks in advance!

aceisace, 2021-01-17:
Closing issue due to lack of feedback. If you feel a need to re-open this issue, please feel free to do so again.

closed 2021-01-17 by aceisace