aceinnolab/Inkycal #153: Feature request: Unique icon per calendar

pietervanharen, 2021-01-12:
With multiple calendars on an InkyCal it not only difficult to keep track of them in the settings.json, but also which meeting belongs to which calendar. It would be great if you could set an icon per calendar, so you could recognise them on the InkyCal screen.

A similar apporch is used in the Calendar module for Magic Mirror: https://github.com/MichMich/MagicMirror/tree/develop/modules/default/calendar

aceisace, 2021-01-13:
@pietervanharen Thanks for your feature request. This seems like a good suggestion but it'll available in the next release, v2.0.1.

aceisace, 2021-02-12:
I'll be closing this issue then as this feature request has been added on the Project page. Once it has been implemented, the respective card will be moved to the implemented branch: [Project page](https://github.com/aceisace/Inkycal/projects/3)
As this feature request has been accepted, I'll be closing this issue. Should you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-12 by aceisace