aceinnolab/Inkycal #162: Monochrome/Bitmap proportional fonts?

surak, 2021-01-21:
Currently, the font used for rendering most of the text is NotoSans. It's readable, but far from looking good in everything that is not that beautiful 9.7" display from #117.

The problem is that TrueType fonts are created to be rendered at any resolution, and with the low resolution and lack of aliasing of e-displays, all kinds of small-scale distortions appear. From the same text being rendered differently from one line to the next, to single-pixel letters in one position being double-pixel right ahead (for example, a letter "l").

The solution for that would be to use a bitmap font. While most bitmap fonts are monospaced and that would look awful, luckily Raspbian has Xorg fonts (check with xfontsel, from the x11-utils package).

The disadvantage is that one might need to install the whole X11 for that. The good thing is that it's all open source and we could copy fonts to this repo.

I think that readability would be greatly increased with the proper typography.

As an example, look how the colons are rendered here with 3 pixels, and how the "ü" is rendered - note the umlaut. This is the default font 12 size which your configuration gives. By the way, the rendering is WAY better with size 14 with this same font - if anything, I would recommend that instead of the current default of 12.

[image]

(Gosh, this thing is working for so long I never noticed how dirty it is)

aceisace, 2021-01-24:
Thanks for your suggestion @surak .
glad to hear your Inkycal has been running for a long time : )
In regards to font-rendering, indeed there still is room for improvements even after coming a long way from non-optimal font rendering, but it will never reach the level of the 9.7" E-Paper display due to the lack of gray-shades. While the 9.7" E-Paper has 16 shades of gray, the 7.5" E-Paper models only have monochrome + pure single colour support. With release 2.0.0 fonts do not bleed anymore, hence the problem with coloured pixels was eleminated, but it's actually pretty difficult further improving the rendering without 1) algorithm to smooth out fonts in monochrome mode 2) using bitmaps font.

While Bitmaps font allow near pixel-perfect readability, there are only a few to choose from, and even those only support a few languages at max. Inkycal uses the NotoSans Font from Google, allowing support for many languages, so finding a replacement would be difficult too.

You are right about the fontsize making a big impact on the readability. Any fontsize above 12 looks pretty decent already.

I'll take this oppurtunity to:
* allow choosing fonts
* allow uploading fonts via the optional web-ui 

Both of these features will be added in the next release. Does this sound like a suitable solution?

markusmo3, 2021-01-25:
Just wanted to chime in and say that i have converted the [slkscr](https://kottke.org/plus/type/silkscreen/index.html) font to a *.pil format which is for pixel perfect rendering.
I also now know how to do this with any other font.
I will post a PR some time in the near future where this font is integrated.

surak, 2021-01-25:
@markusmo3 that is great! Are we allowed to do so with NotoSans?

surak, 2021-01-25:
> glad to hear your Inkycal has been running for a long time : )

You remember it from the early days of the project where I would annoy you to no end :-)

> While Bitmaps font allow near pixel-perfect readability, there are only a few to choose from, and even those only support a few languages at max. Inkycal uses the NotoSans Font from Google, allowing support for many languages, so finding a replacement would be difficult too.

True, but that means that the majority of cases have to be given the same treatment as edge cases (ie different languages) .
 
> I'll take this oppurtunity to:
> * allow choosing fonts
> * allow uploading fonts via the optional web-ui
> Both of these features will be added in the next release. Does this sound like a suitable solution?

All those are great. I can always change myself, as I always did - what I am more thinking about it about what a sane default is. 
My suggestion would be to use by default bitmap font for things like the weather view (which is basically numbers), as well as for the calendar days, as again they are just numbers. For the month's name, weekdays, events, a suitable TrueType/OpenType font is okay, as long as the size is bigger than 12.

What do you think?

aceisace, 2021-01-25:
Yup, I remember the early days of Inykcal, where I had no clue about GIT flow, and most programming formalities. It's thanks to the support, help, suggestions and critics of all users that Inykcal is where it is today : ). Once again, thanks for your help @surak !

Some modules use some sections where space is critical to automatically determine the largest fontsize which can fit in the given space. The weather module does this for almost all text, if not all.

The way the font is selected is by the 'template' module. All Inkycal modules inherit it's parameters, allowing a default value for each. The template uses NotoSans as the default font, hence all modules use the same. It is possible via the software to specify a custom font for each module, but this isn't documented as it shouldn't be used this way. In the next release, how about allowing to choose a font for each module instead of a single one used by all of them directly via the web-ui?

aceisace, 2021-01-25:
@markusmo3 Thanks for the  suggestion. Indeed it's hard picking a bitmap font which fits the exact requirements of an existing software. Your font file does look readable, though I cannot test it myself just yet. 

In the next release, I plan to allow uploading font files via the web-ui. This way, users can specify what font to use, making it more flexible and customizable.

markusmo3, 2021-01-29:
Found a way to disable font aliasing with ttf fonts.
Its included and enabled by default in PR #165

aceisace, 2021-02-12:
Alright then, I've done some testing with the slksrc font as well as some others. The default font has been switched too for the next release, which supports less languages, but a more crisp text on the E-Paper. Here's the link to the currently available fonts:
[Inkycal new fonts](https://github.com/aceisace/Inkycal/tree/feature/web-ui-refactoring/external/fonts).

Some fine-tweaking is still required, but that is outside the scope of this issue. I'll be closing this issue as it has been implemented for the next release. Should you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-12 by aceisace