aceinnolab/Inkycal #163: Authentification Error in ical_parser.py

Kazenwadel, 2021-01-23:
**Describe the bug**
When manually adding username and password in ical_parser to authenticate calendar access, program structure does mention errors. 

**Fix**
Can be fixed by changing import from:

`from urllib.request import urlopen`

to:

`import urllib.request`

and moving `def auth_ical() `before L52.

aceisace, 2021-01-25:
Thanks for the report @Kazenwadel .
The authorisation feature isn't officially implemented yet due to it's difficulty with integrating it in the web-ui. I'll test the code suggestion before pushing it officially in the next release. Thanks for pointing out the necessary changes.

aceisace, 2021-02-01:
@Kazenwadel I'll be closing this issue as this feature request can be tracked on the [Project page](https://github.com/aceisace/Inkycal/projects/3). You can expect this feature to be working as expected in the next official release. If you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-01 by aceisace