aceinnolab/Inkycal #164: Feature Request : E-Mail Module

DerPicknicker, 2021-01-24:
Hey, 

Great project! Is there any chance to read the latest Emails with inkycal? I would suggest that the Subject Title is displayed. 

Is there any way to get it to inkycal? 

Is there a picture for the wiring available..?

Greetings!

worstface, 2021-01-24:
Hello!

There is currently no module for displaying email, but you can create your own modules that add functionality to inkycal.
Gmail e.g. has an API and python library that could possibly be used for that: https://developers.google.com/gmail/api/quickstart/python
So, if you are somewhat proficient in programming/python you could just go ahead and create such module yourself.

Bye!

aceisace, 2021-01-25:
Just as @worstface has mentioned already, there is currently no module which offers this functionality.
I did make an attempt to write such a module in the past, but gave up soon after because of the higher security standards from most major email providors. Authentifaction has become a bit more complicated, making it difficult to write such a module, but you're free to write up your own module for this purpose. 

A few helpful links to get started with third party inkycal modules:
* [Developer docuemntation](https://github.com/aceisace/inkycal_template)
* [Inkycal documentation](https://inkycal.readthedocs.io/en/main/)

In regards to wiring, the white flex cable's metal part should be pointing up. Depending on the label of your E-Paper Display, the orange flex cable in connected with the metal part either down or up. For more specific instructions, please provide the E-Paper label printed in white on the orange flex cable coming out of the E-Paper.

DerPicknicker, 2021-01-25:
Hello again, 

i am not a Python Programmer only C... But i found a way which is a way to go. You get all mails per IMAP and convert it to an RSS feed.. 

What do you think about this idea? The RSS feed should be only local an not hosted to the web. 

https://github.com/cordawyn/rss2imap

DerPicknicker, 2021-01-25:
After I read the Template I cannot understand how to generate a Custom Image...Can I use Cliparts or something like that? 

My Ideas : 
- Over the WebUi you had to set the credentials (My Login alredy works) 
- Also you should be able to select the Folder of the Mailbox .. 
- Displaying Content: 
  - Three Settings:
  	1. Only the Number of unread Messages 
  	2. Displaying a List with 5-10 Emails with Subject and Sender-Adress
  	3. Only the Subject of the latest unread 

Login Works: 

- Imap with SSL works 
- No 2-Factor authorization 

Help Required: 
- How to build an Module : 
  - How to get Data from the Webui into the Code for the Mail-Sync
  - How to use the IMAPLib Correctly
  - I need some Python-Help I am a Python Newbie and not really good with the Python-Technique..

worstface, 2021-01-25:
> Help Required:
> 
> * How to build an Module :
>   
>   * How to get Data from the Webui into the Code for the Mail-Sync
>   * How to use the IMAPLib Correctly
>   * I need some Python-Help I am a Python Newbie and not really good with the Python-Technique..

Have a look at the existing modules: https://github.com/aceisace/Inkycal/tree/main/inkycal/modules
And the modules I've made: https://github.com/worstface/inkycal_modules

Use a simple module as template and start from there.

aceisace, 2021-01-25:
@DerPicknicker If you're on Discord, I can help you with developing your own module if you dm me there. Github issues are a bit slow to respond to and not well suited for code suggestions or dicussions.

The problem with (nearly all) IMAP libraries is that most providers block access to less-secure devices like the rpi (unless using a dedicated software). As most users value the security of two-factor auth. and a more secure login approach, not many users would be willing to go through these lenghts to show email-subjects on their Inkycal.

Maybe you don't need to go through the length of making an official third party module, if you're planning this for yourself, development can be much faster too. You can of course still ask to make it official once you get the hang of it and can provide support for your module.

For the web-ui, this step is simply reading the *required* and *optional* array from the respective module. Since the web-ui does this automaticially, you only need to add these two arrays. Yay! But at the moment, the web-ui hosted at aceisace.eu.pythonanywhere.com contains only currently officially supported modules. There is a copy of this in Inkycal/server. If you run `flask --run` in that folder, you can access your own web-ui via the Raspberry Pi's IP address. Any changes in your own module folder will be reflected in the local web-ui.

Let's start with a more specific approach:
```
Over the WebUi you had to set the credentials (My Login alredy works) -> already explained
Also you should be able to select the Folder of the Mailbox ..       -> make this an optional parameter, like the todoist-module
Displaying Content:
Three Settings:
Only the Number of unread Messages                           -> Check the documentation of the library
Displaying a List with 5-10 Emails with Subject and Sender-Adress  -> Can be implemented with some filtering code
Only the Subject of the latest unread                                         -> Can be implemented with some code
```

```
How to build an Module :
How to get Data from the Webui into the Code for the Mail-Sync
-> This is library specific, so the doc can help you better than myself. I can check it out if I get the time to, for this Dicord is better suited though.

How to use the IMAPLib Correctly
-> Again, I can help you with this via Discord DM, but usually a good library provides some examples on how to use them. Stackoverflow is also pretty useful when it comes to python (and even specific libraries).

I need some Python-Help I am a Python Newbie and not really good with the Python-Technique..
-> Same as above
```

Code example, the data in the arrays is displayed automatically once the module is registered by Inykcal.
```python3
  requires = {
    'api_key': {
      "label":"Please enter your Todoist API-key",
      },
  }

  optional = {
    'project_filter': {
      "label":"Show Todos only from following project (separated by a comma). Leave empty to show "+
              "todos from all projects",
    }
  }
```

aceisace, 2021-02-06:
@DerPicknicker I'm getting back to this via Discord, since further discussion is required. To avoid an overpopulated issue, I'm closing this issue for now. If you feel a need to re-open this issue, please feel free to do so.

closed 2021-02-06 by aceisace