aceinnolab/Inkycal #166: Synology Calendar ICS support

Snake16547, 2021-01-31:
Hi,

Is it possible to add support for synology calendar link. The file it put out validated successfully but the calendar and agendar seems to habe problems with this type of URL.

http://Name:Password@Server/caldav/<Name>/qmrae?export

aceisace, 2021-01-31:
@Snake16547 
It's a bit hard to say what is causing the issue without any additional information about what caused the crash. For starters, please paste the last ~50 lines of the Inykcal log file. Then I can help troubleshoot what caused this issue.

For instructions on how to check the log file, see here: [Check log file](https://github.com/aceisace/Inkycal/wiki#debugging-help)

Snake16547, 2021-01-31:
Here you go:

`31-01-2021 01:08:18 | main |  ERROR: Exception in module 2
Traceback (most recent call last):
  File "/usr/lib/python3.7/urllib/request.py", line 1324, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.7/http/client.py", line 1244, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1290, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1239, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 966, in send
    self.connect()
  File "/usr/lib/python3.7/http/client.py", line 1399, in connect
    super().connect()
  File "/usr/lib/python3.7/http/client.py", line 938, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.7/socket.py", line 707, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/lib/python3.7/socket.py", line 748, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 279, in run
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_calendar.py", line 242, in generate_image
    parser.load_url(self.ical_urls)
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 55, in load_url
    for _ in url]
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 55, in <listcomp>
    for _ in url]
  File "/usr/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/usr/lib/python3.7/urllib/request.py", line 543, in _open
    '_open', req)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 1367, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/usr/lib/python3.7/urllib/request.py", line 1326, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno -2] Name or service not known>
31-01-2021 01:08:18 | inkycal_agenda |  INFO: Image size: (460, 177)
31-01-2021 01:08:18 | main |  ERROR: Exception in module 3
Traceback (most recent call last):
  File "/usr/lib/python3.7/urllib/request.py", line 1324, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.7/http/client.py", line 1244, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1290, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1239, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 966, in send
    self.connect()
  File "/usr/lib/python3.7/http/client.py", line 1399, in connect
    super().connect()
  File "/usr/lib/python3.7/http/client.py", line 938, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.7/socket.py", line 707, in create_connection
[... 89 lines cut ...]
urllib.error.URLError: <urlopen error [Errno -2] Name or service not known>
31-01-2021 01:16:19 | inkycal_agenda |  INFO: Image size: (460, 177)
31-01-2021 01:16:19 | main |  ERROR: Exception in module 3
Traceback (most recent call last):
  File "/usr/lib/python3.7/urllib/request.py", line 1324, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/usr/lib/python3.7/http/client.py", line 1244, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1290, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1239, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 966, in send
    self.connect()
  File "/usr/lib/python3.7/http/client.py", line 1399, in connect
    super().connect()
  File "/usr/lib/python3.7/http/client.py", line 938, in connect
    (self.host,self.port), self.timeout, self.source_address)
  File "/usr/lib/python3.7/socket.py", line 707, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
  File "/usr/lib/python3.7/socket.py", line 748, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -2] Name or service not known`

aceisace, 2021-01-31:
@Snake16547 Looks like further investigation is required. The errors in question is this one:
`urllib.error.URLError: <urlopen error [Errno -2] Name or service not known>`
and 
`socket.gaierror: [Errno -2] Name or service not known`

It seems to be a problem accesing the URL, but it's hard to say where exactly the problem lies. It could be related to proxy, certificates or even with the link itself.

You could dig deeper into this by looking up these two errors or provide a sample link here fortesting purposes. But it might be best to use a better known provider, like Google, Yahoo etc. since they are guaranteed to work with Inkycal.

Snake16547, 2021-01-31:
gotcha switching right now to google calendar - when is a option for a different icon in the agendar possible? Would like to have a candle icon for my birthday calendar

aceisace, 2021-01-31:
Google is probably your best option in terms of stability and uptime :)

Icons are a bit hard to implement, because python uses unicde symbols by default -> `u/f035`. This is not user-friendly at all, but recently I found a way to create a dictionary that maps the icon name to the symbol.

As this repo follows Git-Flow, apart from hotfixes, I cannot push code for new request, only bugfixes. Due to this, this feature will be available in the ext official release.

aceisace, 2021-02-06:
I'm closing this issue as the purpose has been resolved. For further inquiries, please note that Inkycal as of yet (v2.0.0) does not support iCalendars with authentification, although a function does exist for this purpose. The reason behind this is the complexity for using that with the web-ui. Until a proper approach has been found, this feature is on-hold.

aceisace, 2021-02-06:
You can track the status of the font icon support on the [Project's page](https://github.com/aceisace/Inkycal/projects/3). Once again, thanks for your suggestion!

closed 2021-02-06 by aceisace

AndiDad, 2026-09-14:
I’m jumping back into this thread...
I’d like to use Inkycal with Synology Calendar as well.
Synology Calendar supports CalDAV.
Would it be possible to fetch data via CalDAV instead of ICS?
I’d really prefer not to move my calendar to the cloud.