aceinnolab/Inkycal #168: Calendar Error Appeared on February 1

RFNajera, 2021-02-01:
**Describe the bug**
The calendar module was loading fine for days until minutes past midnight today. Then it started throwing an error:

```python
01-02-2021 00:01:50 | main |  ERROR: Exception in module 1
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 279, in run
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_calendar.py", line 263, in generate_image
    grid[days],
KeyError: 31
```
**Desktop (please complete the following information):**
Raspberry pi A+
Raspbian (latest release)
Waveshare 7.5-inch black and white e paper display (v2)

hag2k, 2021-02-02:
Hi,
I had the same problem. I added the following check to  /home/pi/Inkycal/inkycal/modules/inkycal_calendar.py , line 260
It solves the problem for now, maybe there is a better longterm solution.

 ```
for days in days_with_events:
        if days in grid:    <----
          draw_border(
            im_colour,
            grid[days],
            (icon_width, icon_height),
            radius = 6,
            thickness= 1,
            shrinkage = (0.4, 0.2)
            )
```

aceisace, 2021-02-02:
I still can't reproduce this error even after using several different iCalendars. The tests for the calendar module seem to be running fine too. I pushed [this commit](https://github.com/aceisace/Inkycal/commit/948573cb59b7bb86c8075f1a2d721ea49f171942#diff-66e66a31bd89160b6b4ab3d23aa2f1eeb3eb229d3755248c50d6acfd59f7b965) on 22nd Nov 2020, which should have fixed the issue. It's possible you are using a less-up-to date version of the calendar module.

In either case, before I push a hotfix, could you please run the following command:
```bash
cd /home/$USER/Inkycal/inkycal/modules && mv inkycal_calendar.py inkycal_calendar_old.py && wget https://raw.githubusercontent.com/aceisace/Inkycal/main/inkycal/modules/inkycal_calendar.py
```
Then restart Inkycal. If the issue is still not resolved, please let me know. Thanks in advance.

RFNajera, 2021-02-02:
Tried your way, Aceisace, here's the result:

```
pi@inkycalrpi:~ $ python3 inky_run.py
yfinance is not installed! Please install with:
pip3 install yfinance
inkycal_calendar loaded
inkycal_weather loaded
inkycal_feeds loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
generating image(s) for Calendar - Show monthly calendar with events from iCalendars...Error!
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 226, in test
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_calendar.py", line 263, in generate_image
    grid[days],
KeyError: 31

generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...OK!
generating image(s) for RSS / Atom - Display feeds from given RSS/ATOM feeds...OK!
Error/s in modules: 1
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
Date: 2 Feb 21 | Time: 17:33
```

RFNajera, 2021-02-02:
hag2k's fix up there fixed it.
```
pi@inkycalrpi:~ $ python3 inky_run.py
yfinance is not installed! Please install with:
pip3 install yfinance
inkycal_calendar loaded
inkycal_weather loaded
inkycal_feeds loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
generating image(s) for Calendar - Show monthly calendar with events from iCalendars...OK!
generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...OK!
generating image(s) for RSS / Atom - Display feeds from given RSS/ATOM feeds...OK!
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
Date: 2 Feb 21 | Time: 17:43
Generating images for all modules...successful
Initialising..Updating display......Done
Sending E-Paper to deep sleep...Done

No errors since 1 display updates 
program started a minute ago
16 minutes left until next refresh
```

closed 2021-02-02 by aceisace

aceisace, 2021-02-02:
@RFNajera Thanks for testing the the code. Looks like the hotfix was needed. I've implemented the exact suggestion as hag2k . 
Thanks for the report @RFNajera and thanks for the code snippet @hag2k !

PR #170 (2021-02-02): Fix for KeyError: 31
Added a second check for checking if key exists in the dictionary.
This fixes #168 and has been tested by @hag2k.