aceinnolab/Inkycal #172: Installer won't work on a clean raspbian-minimal for lacking git.

surak, 2021-02-16:
I would suggest that the installer checks for git and installs it, if necessary. And stop the installation process if the git clone fails. 

```
$ bash -c "$(curl -sL https://raw.githubusercontent.com/aceisace/Inkycal/main/installer.sh)"
Please select an option from below:
Enter [1] to update Inkycal
Enter [2] to install Inkycal
Enter [3] to uninstall Inkycal
Confirm your selection with [ENTER]
Waiting for input...  2
Cloning Inkycal repo from Github
bash: line 56: git: command not found
Installing Inkycal..
```

worstface, 2021-02-16:
raspberry OS minimal isn't supported at the moment afaik
Some guide how to make it work or making it work with the installer would be great of course.

surak, 2021-02-16:
@worstface it's literally an `sudo apt install git python3-pip` at the beginning of the installer. If the packages are there already, no harm is done either.

aceisace, 2021-02-16:
I can add some instructions in the wiki on how to get it working with Raspberry Pi minimal. The problem is that there are also a number of smaller dependencies, e.g. `libatlas-dev` and a few more of Pillow, which are missing on Raspberry Pi minimal, but are present on Raspberry Pi OS with Desktop.
As the pre-installed packages on Raspberry Pi minimal keep changing in each release, it's hardly worth the effort trying to figure out which ones are exactly missing and then adapting the installer (with respect to GIT-flow).

maweki, 2021-03-14:
Let me get this straight:

It's fine to add dependencies, like https://github.com/aceisace/Inkycal/blob/main/installer.sh#L64
Just not all dependencies? Why not just add all dependencies for Inkycal and let the package manager figure out, what's missing?

maweki, 2021-03-14:
So for dietpi, which is even more minimal than the minimal raspbian I needed to add the following `sudo apt install git python3-pip libtiff5 libopenjp2-7 rpi.gpio libxcb1 libfreetype6-dev libfreetype6-dev python3-pillow; pip3 install spidev setuptools pillow` but I only tried the Openweathermap-module yet.

aceisace, 2021-03-16:
@maweki Thanks for the code suggestion. May I ask you to create a PR for this? 
Looks like I'll have to look into Dietpi to use it as a minimal OS and install missing packages required by Inkycal.
This may take some while since I'm a bit busy with exams these days, but once I get the time to, I'll try re-adding support for Raspbian minimal and inofficial support for Dietpi.

maweki, 2021-03-16:
> May I ask you to create a PR for this?

will do

> I'm a bit busy with exams these days

I know. I have to administer quite a few either.

maweki, 2021-03-16:
What's the policy on the `spidev` dependency? It's included in `epdconfig.py` but not necessary for the `9_in_7.py` display interface.

There's also no hint in the install instructions that this would be a (soft) dependency for most displays. Just add it as a pip-dependency for Inkycal, install it seperately, or don't install it at all (possibly letting the user deal with it)?

aceisace, 2021-03-16:
Added note to 9.7" driver file: https://github.com/aceisace/Inkycal/commit/65f28bea853a429346dbdd49488110cedcf54a4f
Spidev is only required by the 9.7" Display and there is a different version between the pip3 and the package used in Inkycal. The used spidev package is not hosted on Pypi. Installation instructions are shown as soon as the 9.7" Display is run with inkycal.

The dependencies of the core modules of Inkycal are installed by default. On the Zero W, it would be a pain if dependencies had to be installed one by one per selected module.

On second thought, I found a better way to handle this without modifying the installer:
https://github.com/aceisace/Inkycal/commit/a1e281b2656a7136182849f9d24ed740fa7088d2

In short, this additional line has to be run by users with Raspberry OS minimal. Since Pillow is a dependency of Inykcal, I've removed it from your suggested line. That way, sudo is not required and the installer can remain as-is.

Thanks for your code suggestion about the missing packages!

It's getting late over here, so I'll get back to this some other time.

aceisace, 2022-08-27:
Closing this issue as support for venv has been officially implemented in the `feature/venv` branch, which will be merged into main and is scheduled for official release soon

closed 2022-08-27 by aceisace