aceinnolab/Inkycal #175: datetime.max.timestamp and other exceptions during first run

phuntee, 2021-02-26:
**Describe the bug**
I have performed a clean install on a Pi Zero WH and have a 5.83" epaper display attached. I get the following error(s) when I manually run inky_run.py:

pi@rpiz-inky:~ $ python3 inky_run.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/constants.py", line 16, in <module>
    _MAX_TIMESTAMP = datetime.max.timestamp()
OverflowError: timestamp out of range for platform time_t

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "inky_run.py", line 1, in <module>
    from inkycal import Inkycal # Import Inkycal
  File "/home/pi/Inkycal/inkycal/__init__.py", line 5, in <module>
    import inkycal.modules.inkycal_agenda
  File "/home/pi/Inkycal/inkycal/modules/__init__.py", line 1, in <module>
    from .inkycal_agenda import Agenda
  File "/home/pi/Inkycal/inkycal/modules/inkycal_agenda.py", line 10, in <module>
    from inkycal.modules.ical_parser import iCalendar
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 16, in <module>
    import arrow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/__init__.py", line 2, in <module>
    from .api import get, now, utcnow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/api.py", line 12, in <module>
    from arrow.arrow import TZ_EXPR, Arrow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 34, in <module>
    from arrow import formatter, locales, parser, util
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/parser.py", line 26, in <module>
    from arrow.util import next_weekday, normalize_timestamp
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/util.py", line 6, in <module>
    from arrow.constants import (
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/constants.py", line 25, in <module>
    else datetime(2038, 1, 18, 23, 59, 59, 999999).timestamp()
OverflowError: timestamp out of range for platform time_t

**Additional context**
No error or warning messages were observed during install.
I have limited knowledge of Python programming but  can find my way around fairly quickly.

aceisace, 2021-02-26:
Thanks for the report @phuntee !
From the traceback it seems to be a dependency issue related to the arrow package. I'll need your help with this to identify where the issue came from. For this, please try the following:
* Open Terminal and enter `pip3 show arrow`. You'll see some output, please paste the output in your reply. 
* Try to re-install arrow: `pip3 uninstall arrow && pip3 install arrow`. Then reboot with `sudo reboot`
If after trying the above suggestion, it didn't help after rebooting, most likely a hotfix is required. If this is the case, I'll dig deeper into this issue.

phuntee, 2021-02-26:
Thank you very much for responding! I ran the show arrow command and rebooted but it hasn't helped. Here is the output from the pip3 show arrow command:

pi@rpiz-inky:~/Inkycal/logs $ pip3 show arrow
Name: arrow
Version: 1.0.0
Summary: Better dates & times for Python
Home-page: https://arrow.readthedocs.io
Author: Chris Smith
Author-email: crsmithdev@gmail.com
License: Apache 2.0
Location: /home/pi/.local/lib/python3.7/site-packages
Requires: typing-extensions, python-dateutil
Required-by: inkycal
pi@rpiz-inky:~/Inkycal/logs $

********************************

pi@rpiz-inky:~ $ python3 inky_run.py
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/constants.py", line 16, in <module>
    _MAX_TIMESTAMP = datetime.max.timestamp()
OverflowError: timestamp out of range for platform time_t

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "inky_run.py", line 1, in <module>
    from inkycal import Inkycal # Import Inkycal
  File "/home/pi/Inkycal/inkycal/__init__.py", line 5, in <module>
    import inkycal.modules.inkycal_agenda
  File "/home/pi/Inkycal/inkycal/modules/__init__.py", line 1, in <module>
    from .inkycal_agenda import Agenda
  File "/home/pi/Inkycal/inkycal/modules/inkycal_agenda.py", line 10, in <module>
    from inkycal.modules.ical_parser import iCalendar
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 16, in <module>
    import arrow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/__init__.py", line 2, in <module>
    from .api import get, now, utcnow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/api.py", line 12, in <module>
    from arrow.arrow import TZ_EXPR, Arrow
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 34, in <module>
    from arrow import formatter, locales, parser, util
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/parser.py", line 26, in <module>
    from arrow.util import next_weekday, normalize_timestamp
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/util.py", line 6, in <module>
    from arrow.constants import (
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/constants.py", line 25, in <module>
    else datetime(2038, 1, 18, 23, 59, 59, 999999).timestamp()
OverflowError: timestamp out of range for platform time_t

aceisace, 2021-02-26:
Thanks for getting back so fast. The output now explains a few things.
This release of arrow (v1.0.0) is pretty new and therefore not ready to be used with Inkycal. Inkycal used to get the latest version of arrow, but a fixed version in now required (v0.17.0).
I'll push a hotfix to this in the main branch. To get Inkycal working again, please try the following now:
`pip3 uninstall arrow && pip3 install arrow==0.17.0`
This should hopefully fix the issue, after rebooting. Please let me know if it worked for you :)

phuntee, 2021-02-26:
Seems to have worked. But I now have other errors related to image generation. Here are the new errors:

pi@rpiz-inky:~ $ python3 inky_run.py
inkycal_weather loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_5_in_83_colour
generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...Error!
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 226, in test
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 460, in generate_image
    '\uf053')
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 198, in draw_icon
    space = Image.new('RGBA', (box_width, box_height))
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2614, in new
    _check_size(size)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2593, in _check_size
    raise ValueError("Width and height must be >= 0")
ValueError: Width and height must be >= 0

Error/s in modules: 1
Inkycal version: v2.0.0
Selected E-paper display: epd_5_in_83_colour
Date: 26 Feb 21 | Time: 14:07
Generating images for all modules...error!
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 279, in run
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 460, in generate_image
    '\uf053')
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 198, in draw_icon
    space = Image.new('RGBA', (box_width, box_height))
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2614, in new
    _check_size(size)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2593, in _check_size
    raise ValueError("Width and height must be >= 0")
ValueError: Width and height must be >= 0

26-02-2021 14:07:05 | main |  ERROR: Exception in module 1
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 279, in run
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 460, in generate_image
    '\uf053')
  File "/home/pi/Inkycal/inkycal/modules/inkycal_weather.py", line 198, in draw_icon
    space = Image.new('RGBA', (box_width, box_height))
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2614, in new
    _check_size(size)
  File "/home/pi/.local/lib/python3.7/site-packages/PIL/Image.py", line 2593, in _check_size
    raise ValueError("Width and height must be >= 0")
ValueError: Width and height must be >= 0
error/s in modules: 1

aceisace, 2021-02-26:
Glad to hear it's been resolved. I've just now pushed the hotfix and this issue should no longer be happening in future installations. Once again, apologies for the inconvenience and thanks for the report!

In regards to the next error, it seems as though your weather module has a too small height to display anything reasonable. Can you paste your settings file and remove any sensitive data from there? That will hopefully clear up things further.

phuntee, 2021-02-26:
Hi, thank you again for the quick response. You shouldn't apologise, this is art in development!

Here is the settings.json file:

{
    "model": "epd_5_in_83_colour",
    "update_interval": 30,
    "orientation": 180,
    "info_section": true,
    "info_section_height": 30,
    "calibration_hours": [
        0,
        12,
        18
    ],
    "modules": [
        {
            "position": 1,
            "name": "Weather",
            "config": {
                "size": [
                    448,
                    570
                ],
                "api_key": "<redacted>",
                "location": "<redacted>",
                "round_temperature": false,
                "round_windspeed": false,
                "forecast_interval": "daily",
                "units": "metric",
                "hour_format": "12",
                "use_beaufort": true,
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "en"
            }
        }
    ]

phuntee, 2021-02-26:
I have removed the OpenWeather API key and the location, rest of the output is exact.

aceisace, 2021-02-26:
Thanks for the settings file. I figured out the issue. As the only used module is the weather module, it has difficulties using the full screen (since Inkycal by default stretches a single module to full screen). If you add a second module, e.g. like the simple Jokes Module, your settings file will allocate less space to the weather module, which will fix this issue.

You can also use a different layout if you like. If you need a starting point, a well-looking configuration would be to use:
* weather module as module 1: height -> 15
* Calendar module as module 2: height -> 60
* RSS/Jokes Module as module 3: height ->25

Once you have a working settings file, you can adjust things as you see fit by adding/removing more modules and changing heights : )

phuntee, 2021-02-26:
Cool, let me give that a try now.

phuntee, 2021-02-26:
So, I tried something but have a question. How much time does it take to render the output onto the display? My output is stuck at the following for 8 minutes and nothing has been sent to the display:

pi@rpiz-inky:~ $ python3 inky_run.py
inkycal_weather loaded
inkycal_jokes loaded
inkycal_feeds loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_5_in_83_colour
generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...OK!
generating image(s) for iCanHazDad API - grab a random joke from icanhazdad api...OK!
generating image(s) for RSS / Atom - Display feeds from given RSS/ATOM feeds...OK!
Inkycal version: v2.0.0
Selected E-paper display: epd_5_in_83_colour
Date: 26 Feb 21 | Time: 15:50
Generating images for all modules...successful

Log file for this timestamp:

26-02-2021 15:50:08 | main |  INFO: name : Weather size : 448x85 px
26-02-2021 15:50:08 | main |  INFO: name : Jokes size : 448x142 px
26-02-2021 15:50:08 | main |  INFO: name : Feeds size : 448x342 px
26-02-2021 15:50:08 | inkycal_weather |  INFO: Image size: (428, 65)
26-02-2021 15:50:09 | inkycal_weather |  INFO: Connection test passed
26-02-2021 15:50:16 | inkycal_jokes |  INFO: image size: 428 x 122 px
26-02-2021 15:50:17 | inkycal_jokes |  INFO: Connection test passed
26-02-2021 15:50:18 | inkycal_feeds |  INFO: Image size: (428, 322)
26-02-2021 15:50:19 | inkycal_feeds |  INFO: Connection test passed
26-02-2021 15:50:25 | inkycal_weather |  INFO: Image size: (428, 65)
26-02-2021 15:50:26 | inkycal_weather |  INFO: Connection test passed
26-02-2021 15:50:29 | inkycal_jokes |  INFO: image size: 428 x 122 px
26-02-2021 15:50:34 | inkycal_jokes |  INFO: Connection test passed
26-02-2021 15:50:35 | inkycal_feeds |  INFO: Image size: (428, 322)
26-02-2021 15:50:35 | inkycal_feeds |  INFO: Connection test passed
************************

Unsure where the issue is.
Thank you for spending time on this!

aceisace, 2021-02-26:
Glad to hear you got a bit further now :)
Usually, even on the Zero W, it's rare for the specified modules to take more than 3 minutes to generate. From the log it seems as though the images have been generated correctly too. It's a bit hard finding the exact cause of this (incorrect driver, wiring issue). But for starters, the following questions might help:
* Does your display support colour or not?
* Have you managed to get waveshare's example codes working before or now with the same setup?
* Have you enabled SPI according to the instructions on the README?
* Have you double checked your wiring?

Once you answer these questions, the next step will hopefully be clearer. It's getting late here, so I'll get back tomorrow.

phuntee, 2021-02-28:
Thanks for replying.

1. Display I ordered is colour, RBG 5.83" 
2. I'll try Waveshare's example code, I haven't testing the display with that.
3. SPI is enabled.
4. I've checked the wiring a few times and seems okay. I'll remove the HAT and the ribbons and put them back together once more to be sure. I'll then run Waveshare's sample code to test first.
Will post updates if my weekend doesn't go busy!

aceisace, 2021-03-01:
@phuntee I think a display test at this stage might be helpful to rule out issues related to wiring, drivers and SPI.
For this, please download the image below and copy it to the Raspberry Pi in `/home/pi`. Rename this image to `pikachu.png`
[image]

```python
from inkycal import Display
from PIL import Image

image = Image.open("/home/pi/pikachu.png")
display = Display("epd_5_in_83_colour")
display.render(image)
```
If the image is shown on the display, any issues related to rendering, wiring and drivers can be ruled out. If this isn't the case, please paste the traceback.

phuntee, 2021-03-01:
Hey,

Thank you for replying. I got it working today. Waveshare is now shipping v2 of these displays. So I had to change the display driver to Waveshare's epd5in83b_V2.py. I think InkyCal was originally set to use epd5in83.py. Results:

[image]

[image]

I will now work on improving the way weather is displayed. I want one large section showing current weather and small ones below in a line showing forecast.

Thank you again for all the help and information. This is an excellent project and you've done well!

This thread can now be closed.

aceisace, 2021-03-03:
@phuntee Glad to hear you got Inkycal working on your display! Apologies for the inconvenience with the driver files. I had no clue Waveshare was shipping v2 of the 5.83" variant, but I'll add support for this soon enough. Thank you for your patience. I've added a note about this in the project page and will work on this soon.

Oh, that's a nice photo frame! Do you want to share your builf on the [hall of fame page](https://github.com/aceisace/Inkycal/wiki/hall-of-fame)? 

Looks like you also got the layout figured out : ). You can add several modules with Inkycal, even duplicate ones are fine. You can specify module1 as weather with hourly forecast while specfying a second weather module (e.g. module 4) with daily forecast. Unfortunately changing the layout within a module is a bit more difficult and only suggested for expierenced users.

phuntee, 2021-03-03:
Thank you! Sure, I would love to share my build on the hall fame page, I'll send you an email soon!

closed 2021-03-03 by phuntee