aceinnolab/Inkycal #180: Uninstall Script not clearing crontab

rtyoung000, 2021-03-16:
**Describe the bug**
While troubleshooting inkycal to determine what was wrong with the screen, I uninstalled and re-installed using the automated script (very complete by the way nice work!). After second install I restarted and when I ran ps aux | grep python I noticed there were two inky_run.py running. The fix was simple I just jumped into crontab -e and deleted one of the lines.

**To Reproduce**
Steps to reproduce the behavior:
1. run install script -> select new install option
2. reboot
3. run install script -> select uninstall option
4. reboot
5. run install script -> select new install option
6. crontab -e

**Expected behavior**
You should see 2 inkycal entries

aceisace, 2021-03-16:
Thanks for pointing this out, @rtyoung000 . I'll implement a fix for this today/tomorrow.

aceisace, 2021-03-16:
@rtyoung000 
OK, I've implemented a hotfix for this issue. It should now work as expected: https://github.com/aceisace/Inkycal/commit/e4e0493f0e079396be011a03964bca46a6f0033b

Simply running `crontab -r` fixed it. This reverts the crontab file to it's original state.
Closing this issue as it has been resolved.

Should you feel a need to re-open this issue, please feel free to do so.

closed 2021-03-16 by aceisace