aceinnolab/Inkycal #182: Suggestion: Add Systemd service instead of cron @reboot

maweki, 2021-03-16:
(I will open a PR if this is wanted, no work from you necessary)

Currently InkyCal adds itself to cron with the `@reboot` rule. That's not very resilient against crashing or loss of network connection, etc.

Inkycal can easily be added as a systemd simple service which allows it to be started only after network connection has been established, and also allows it to be automatically restarted once it crashes. No code changes in the actual application are necessary for that. We only need to put a service file at a specific place and enable the service.

aceisace, 2021-03-16:
Yes, you are right, crontab is fast to setup but hard to monitor.
In the BETA OS of Inkycal, I already implemented a more suitable option, `supervisor`. This allows automated logging, automated restarts of Inkycal and can be configured within a few lines. Best of all, it already was implemented in earlier releases, so I'm well familiar with it :)

Systemd is an option too, but it's more complicated to setup and more prone to errors. As Inkycal aims to be as user-friendly as possible, it makes more sense to rely on `supervisor` in case users wish to customize the behaviour without needing to learn  much about systemd.

For reference, a simple supervisor conf file, handling logging, retry counter, user etc.
```
[program:Inkycal]
command = python3 /....inkyrun.py
user = pi
redirect_stderr=true
stdout_logfile = /..Inkycal/logs.log
stdout_logfile_maxbytes = 1MB
startsecs = 10
startretries=3
autorestart=false
```

maweki, 2021-03-16:
Anything that's more resilient is fine. Just as a reference, a unit file would be as simple as

```
[Unit]
Description=Inkycal
After=network.target
[Service]
Type=simple
User=ouruser
WorkingDirectory=inkycaldirectory
ExecStart=python3 pathtoinkycal
Restart=always
RestartSec=30
[Install]
WantedBy=multi-user.target
```

aceisace, 2021-03-16:
Yup, you're right, crontab has it's limitations. I've already implemented supervisor in the BETA OS, so I'll add official support for this in the coming release.
Thanks for the infor about systemd :)

aceisace, 2021-11-11:
Closing this issue since the suggestion has been taken on for the next official release. If you feel a need to re-open this issue, please feel free to do so.

closed 2021-11-11 by aceisace