aceinnolab/Inkycal #183: Module list doesn't fully respect position attribute

maweki, 2021-03-16:
As I orchestrate my whole home-automation system through ansible, I am writing my `settings.json` file by hand. As it turns out, if the order of the items in the `modules` list does not match the `position` attribute, you get strange behaviours.

The position attribute should either be removed and given implicitly through the actual position,
or the order of the list should not matter and it should work independent of the order of modules, just given the positions. I'm fine with either and am happy to provide a PR for either solution.

aceisace, 2021-03-16:
Thanks for the report. Indeed, this is a known issue, but not one that is common and needs a hotfix right away.
This is just a remnant of the transition from Inykcal v1.x.x -> 2.0.0 and it's more complicated than a single file (web-ui, main.py, settings_parser, settings_generator etc.) but a fix for this has been applied in the BETA already.

You may want to join the [Inkycal Discord Server](https://discord.gg/sHYKeSM) and discuss features, bugs etc. with other users/developers directly :)

Closing this issue as a fix for this has been applied in the BETA and is sheduled for the new release.

closed 2021-03-16 by aceisace

maweki, 2021-03-16:
> but a fix for this has been applied in the BETA already

Doesn't work in the `development` branch. Is there somewhere else where the BETA is, that is not the `development` branch?

aceisace, 2021-03-16:
See here :)
https://github.com/aceisace/Inkycal/blob/a130a7007d3fbf0f08895df2babfcc66b5ba4217/inkycal_ui/app/routes.py#L190
And yes, the branch with the latest developments is feature/web-ui-refactoring