aceinnolab/Inkycal #188: Timezone error with Outlook Calendar

thorok73, 2021-04-12:
**Describe the bug**
Module 2 in my config is the calender module. Since Wednesday last week it occasionally does not update the calendar anymore. 
The calendar I use in inkycal since some months already is an outlook.com calendarfrom : https://outlook.live.com/owa/calendar/...

When running inkcal manually I get the following module error:

Error/s in modules: 2
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
Date: 12 Apr 21 | Time: 13:31
Generating images for all modules...error!
Traceback (most recent call last):
  File "/home/pi/Inkycal/inkycal/main.py", line 279, in run
    black,colour=module.generate_image()
  File "/home/pi/Inkycal/inkycal/modules/inkycal_calendar.py", line 251, in generate_image
    month_events = parser.get_events(month_start, month_end, self.timezone)
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 148, in get_events
    if events: self.parsed_events += list(events)
  File "/home/pi/Inkycal/inkycal/modules/ical_parser.py", line 144, in <genexpr>
    } for ical in recurring_events for events in ical)
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/api.py", line 19, in get
    return _factory.get(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/factory.py", line 187, in get
    return self.type.fromdatetime(arg)
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 264, in fromdatetime
    fold=getattr(dt, "fold", 0),
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/arrow.py", line 98, in __init__
    tzinfo = parser.TzinfoParser.parse(tzinfo.zone)
  File "/home/pi/.local/lib/python3.7/site-packages/arrow/parser.py", line 593, in parse
    'Could not parse timezone expression "{}"'.format(tzinfo_string)
arrow.parser.ParserError: Could not parse timezone expression "tzone://Microsoft/Custom"

12-04-2021 13:31:48 | main |  ERROR: Exception in module 2

Until last Wednesday it worked perfectly: But now the update fails.

aceisace, 2021-04-12:
Hi there and thanks for reporting this issue @thorok73 .
Inkycal doesn't officially support Outlook calendars, since they are a bit different than the officially supported .ics calendars and keep changing over time.

From the error log, it seems that the timezone is not set correctly in your Outlook Calendar:
```bash
"tzone://Microsoft/Custom"
```
This error is raised because the timezone could not be parsed by the system. You might have some better luck if you try changing the timezone of your Calendar to a specific value that Inkycal understands, else I'd suggest syncronising your outlook calendar with e.g. Google or Yahoo. The link will change, but any addition/deletion will be reflected almost instantly.

thorok73, 2021-04-13:
Thank you @aceisace for your support. Unfortunately it's a family calendar and I cannot easily change it to another calendar provider.

Where in inkycal could I change the which variable and set it to "Europe/Berlin" manually instead of trying to read this value from outlook.com?

aceisace, 2021-04-13:
Looks like you've made it :)
Another option might be to replace the timezone in the iCalendar module of Inkycal, right before parsing, but it would require some regex to find the correct line. 
This solution has been added to the [wiki](https://github.com/aceisace/Inkycal/wiki/Outlook-Support) now, thanks for the solution @thorok73

aceisace, 2021-04-13:
I'll be closing this issue since it has been resolved. Thanks for getting back with the fix! 💯 

Should you feel a need to re-open this issue, please feel free to do so. Thanks in advance.

closed 2021-04-13 by aceisace