aceinnolab/Inkycal #198: Install doen't work

BenniBlindflug, 2021-09-05:
I use the inkycal for years, until August. I get the error "31 not in list". So I try to update the pi and with the installer.py an update. That doesn't work. So I try a new installation "raspi OS" and "Inkycal". The settings.json is in /boot. The inkcal.log is empty.  Raspi is reacheable via SSH an has a Wifi connection. It is a raspi zero W with a 8MB SD. So, I have no idea what went wrong.
 
If I run the "/home/pi/inky_run.py", I get an error (see last line)

pi@EInkRaspi:~ $ python3 /home/pi/inky_run.py
Traceback (most recent call last):
  File "/home/pi/inky_run.py", line 3, in <module>
    inky = Inkycal(render = True) # Initialise Inkycal
  File "/home/pi/Inkycal/inkycal/main.py", line 118, in __init__
    settings = json.load(settings_file)
  File "/usr/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.7/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting ':' delimiter: line 46 column 17 (char 1491)

worstface, 2021-09-05:
Could be this bug: https://github.com/ranaroussi/yfinance/issues/764

Could you check your yfinance version? it should be 0.1.63

aceisace, 2021-09-06:
@BenniBlindflug 
Hi, and thanks for your interest in Inkycal! Nice to hear you've been a long-term user of Inkycal :)

In regards to the error you are getting, it seems to be related to the settings file itself, i.e. the format. The hint in your traceback is line `46`. Could you please copy and paste line 46 of your settings file in a comment here?

BenniBlindflug, 2021-09-06:
line 46:            "ical_files": null,

aceisace, 2021-09-06:
@BenniBlindflug 
Hmm, this line doesn't say much, it's pretty sure a syntax error. Could you paste your settings.json file here after removing sensitive data? Thanks in advance.

BenniBlindflug, 2021-09-08:
Hi, here are the "settings.json". The file was created from the web site, instead the sensitiv data. The URLs were fitted after creation. Many thanks for caretaking!
------------------------------
{
    "model": "epd_7_in_5_v3",
    "update_interval": 60,
    "orientation": 0,
    "info_section": false,
    "info_section_height": null,
    "calibration_hours": [
        0,
        12,
        18
    ],
    "modules": [
        {
            "position": 1,
            "name": "Weather",
            "config": {
                "size": [
                    528,
                    146
                ],
                "api_key": "api_key",
                "location": "wohnort, DE",
                "round_temperature": true,
                "round_windspeed": true,
                "forecast_interval": "daily",
                "units": "metric",
                "hour_format": "24",
                "use_beaufort": true,
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "de"
            }
        },
        {
            "position": 2,
            "name": "Calendar",
            "config": {
                "size": [
                    528,
                    440
                ],
                "week_starts_on": "Monday",
                "show_events": true,
                "ical_urls": "url_calendar 1","url_calendar 2"
                "ical_files": null,
                "date_format": "D MMM",
                "time_format": "HH:mm",
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "de"
            }
        },
        {
            "position": 3,
            "name": "Agenda",
            "config": {
                "size": [
                    528,
                    146
                ],
                "ical_urls": "url_calendar 1","url_calendar 2"
                "ical_files": null,
                "date_format": "ddd D MMM",
                "time_format": "HH:mm",
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "de"
            }
        },
        {
            "position": 4,
            "name": "Feeds",
            "config": {
                "size": [
                    528,
                    146
                ],
                "feed_urls": "https://rss.focus.de/fol/XML/rss_folnews_eilmeldungen.xml",
                "shuffle_feeds": true,
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "de"
            }
        }
    ]
}
-----------------------------------------

BenniBlindflug, 2021-10-14:
Hi, I have reinstall the system and fixed the "key-values" and... it looks better:
Log: 
14-10-2021 19:01:00 | inkycal_weather |  INFO: Image size: (460, 20)
14-10-2021 19:01:00 | inkycal_weather |  INFO: Connection test passed
14-10-2021 19:01:02 | inkycal_calendar |  INFO: Image size: (460, 580)
14-10-2021 19:03:00 | ical_parser |  INFO: loaded iCalendars from URLs
14-10-2021 19:05:02 | inkycal_agenda |  INFO: Image size: (460, 60)
14-10-2021 19:06:55 | ical_parser |  INFO: loaded iCalendars from URLs
14-10-2021 19:07:53 | inkycal_agenda |  INFO: Managed to parse events from urls
14-10-2021 19:07:53 | inkycal_feeds |  INFO: Image size: (460, 60)
14-10-2021 19:07:53 | inkycal_feeds |  INFO: Connection test passed
14-10-2021 19:07:57 | inky_image |  INFO: merged given images into one
14-10-2021 20:01:00 | inkycal_weather |  INFO: Image size: (460, 20)
14-10-2021 20:01:01 | inkycal_weather |  INFO: Connection test passed
14-10-2021 20:01:03 | inkycal_calendar |  INFO: Image size: (460, 580)
14-10-2021 20:03:01 | ical_parser |  INFO: loaded iCalendars from URLs
14-10-2021 20:05:02 | inkycal_agenda |  INFO: Image size: (460, 60)
14-10-2021 20:06:55 | ical_parser |  INFO: loaded iCalendars from URLs
14-10-2021 20:07:53 | inkycal_agenda |  INFO: Managed to parse events from urls
14-10-2021 20:07:53 | inkycal_feeds |  INFO: Image size: (460, 60)
14-10-2021 20:07:54 | inkycal_feeds |  INFO: Connection test passed
14-10-2021 20:07:57 | inky_image |  INFO: merged given images into one

but... there is still the old image on the display.  The new image is not visible :-(