aceinnolab/Inkycal #200: Problem with feedparser

gonzo2474, 2021-11-10:
Hello,

when I start Inkycal with "python3 /home/pi/inky_run.py &", there is the following error

`  File "/home/pi/.local/lib/python3.9/site-packages/feedparser.py", line 93, in <module>
    _base64decode = getattr(base64, 'decodebytes', base64.decodestring)
AttributeError: module 'base64' has no attribute 'decodestring'`

In line 93 of the file, there is written:
`    # Python 3.1 deprecates decodestring in favor of decodebytes
    _base64decode = getattr(base64, 'decodebytes', base64.decodestring)`

It seems, there is a problem, using the old version of feedparser.py (5.2.1), (actual version is 6.0.8) and actual version of python 3.9.2. But Inkycal needs version 5.2.1 of feedparser. Using version 6.0.8 of feedparser does not work.

aceisace, 2021-11-11:
@gonzo2474 Thanks for the report about the bug. Yes, it seems that there is an issue with python 3.9 and feedparser from the log. I'll push a dependency hotfix into the current release after testing a few things.

closed 2021-11-11 by aceisace

aceisace, 2021-11-11:
@gonzo2474 Okay, the hotfix has been implemented. The tests are now working fine and run on python 3.9.8 instead of 3.7.2. Feedparser needs to be on v6.0.8 for everything to work properly.

If you're looking for a maual fix, you can try this:
`pip3 uninstall feedparser && pip3 install feedparser==6.0.8`

Then reboot (`sudo reboot`) to apply changes and try running Inkycal again. The RSS module should then be working as expected.

I'll be closing this issue as the hotfix has been implemented. If you feel a need to re-open this issue, please feel free to do so

gonzo2474, 2021-11-12:
Thank you. It seems to work, because there is no errormessage to be seen. 
But the display is still clear and white. I will check the wires, maybe there is a problem with the display, the HAT or the connectors. I'll see.
But thank you again for solving the issue

aceisace, 2021-11-12:
@gonzo2474 You‘re most welcome 😄 
You can post the wiring issue at our Discord server, there it‘s easier to debug and will keep this issue clean. Most likely it‘s a small issue with selecting the correct drivers. 

But glad to hear the fix now works :)

PR #202 (2021-11-11): Hotfix/feedparser dependency
This fixes #200 (issue with feedparser dependency) running on the latest Raspberry Pi OS image. Checks for this commit have passed, merging.