aceinnolab/Inkycal #209: high CPU load with new installation

ralf-neugebauer, 2022-02-01:
I have a very high CPU load when I run InkyCal.

I had to install the whole OS and InkyCal on a raspberry W.
It was working on an old installation, but when I take the old settings.json, phyton just take 80% of the CPU.

my settings.json:
{
    "model": "epd_7_in_5_v2_colour",
    "update_interval": 60,
    "orientation": 0,
    "info_section": false,
    "info_section_height": null,
    "calibration_hours": [
        0,
        12,
        18
    ],
    "modules": [
        {
            "position": 1,
            "name": "Calendar",
            "config": {
                "size": [
                    528,
                    880
                ],
                "week_starts_on": "Monday",
                "show_events": true,
                "ical_urls": "https://calendar.google.com/calendar/ical/****",
                "ical_files": null,
                "date_format": "D MMM",
                "time_format": "HH:mm",
                "padding_x": 10,
                "padding_y": 10,
                "fontsize": 12,
                "language": "de"
            }
        }
    ]
}

uname -a:
Linux raspberrypi 5.10.92+ #1514 Mon Jan 17 17:35:21 GMT 2022 armv6l GNU/Linux

python -V
Python 3.9.2

Thank you in advance.

aceisace, 2022-02-06:
Hi there @ralf-neugebauer and thanks for the `settings.json` file.
There can be several reasons behind a CPU usage, often related to incorrect configurations. Hence, the settings file is not enough to help debug this issue. Can you paste the output of `Inkycal/logs/inkycal.log` please? Thanks in advance

ralf-neugebauer, 2022-02-10:
Dear aceisace,

Thank you for your response.

But there is not much:
10-02-2022 16:19:11 | main |  INFO: name : Calendar size : 528x880 px
10-02-2022 16:19:11 | inkycal_calendar |  INFO: Image size: (508, 860)
10-02-2022 16:19:23 | ical_parser |  INFO: loaded iCalendars from URLs
10-02-2022 16:19:25 | inkycal_calendar |  INFO: Image size: (508, 860)
10-02-2022 16:19:31 | ical_parser |  INFO: loaded iCalendars from URLs

Thank you in advance.

aceisace, 2022-04-01:
Apologies for the late reply. It seems that Inkycal is running fine. What is the content in the crontab file?

aceisace, 2022-04-14:
I'm assuming there were multiple instances running Inkycal via the crontab file. Closing this issue as it has become stale. Should you feel a need to re-open this issue, please feel free to do so

closed 2022-04-14 by aceisace