aceinnolab/Inkycal #217: Add support for Waveshare 2.7 inch and others

KodeMunkie, 2022-04-13:
I'd like to use this with my Waveshare 2.7 inch, but requirements say 4.2 is the minimum.

I don't mind not having certain features if the display is too small for some of the modules - anything is better than nothing!

aceisace, 2022-04-14:
The 2.7" displays are smaller than the Rpi itself. As they do not have a high resolution and not enough greyscales  to display text reasonably well, hence it would ruin the point of a fully-fledged dashboard.
Do you have a special use case for these smaller displays running Inkycal?

KodeMunkie, 2022-04-14:
No special use case. I have will shortly have a 3D printed Pi case and a 2.7 inch display on order to put in it.
The display will be used to show metrics from the Pi, weather metrics etc on rotation and possibly custom data long term. This project seemed to achieve a similar goal and a framework seemed like a good start to determine the feasibility and display's usefulness to do this without writing code as I had assumed that a mobile view, i.e. akin to cell phone usage, or partial data view, could be achieved with minimal effort or tweaking of the driver folder.

bjinse, 2022-06-15:
I thought about using a 2.13 inch display to display some temperature readings within my camper van (fridge / behind the fridge / outside van / inside van and voltage reading of the second car battery. Guess I will borrow some code, or create support for this display myself into inkycal.

aceisace, 2022-08-27:
Closing this issue as it has become stale. Please understand that Inkycal is targeted for Display sizes larger than 4.2", which is the minimum recommended size for Inkycal. Even if I do add support for the 2.9", as I do not have or use the 2.9", it would be hard maintaining it's support in future releases.

I can however help you with a simple script for the Raspi that renders some basic text and sensor values. If you are interested, please send me a mail or shout out on our Discord server.

Using a Raspberry Pi for reading sensor data is a bit overkill of hardware. For that purpose I'd recommend using an esp32, which is much better suited for this task due to low-power, deepsleep and faster rendering.
For that purpose, I'd recommend checking out this repo: https://github.com/ZinggJM/GxEPD

However, as Inkycal is open-source, you are most welcome to use parts of the Inkycal code to help you with this purpose

closed 2022-08-27 by aceisace