aceinnolab/Inkycal #219: Add support for Inky pHAT and wHAT

CapnBio, 2022-04-28:
Release version = 2.0.0
Expected behavior = e-ink display to update with modules
Actual behavior = upon start up OS works fine, no update on display
Steps to reproduce the behavior = before installation input all things needed for installation and inky wHAT is already connected

I am trying out this code today and it is not working out of the box after installation with the Inky wHAT with a 4.2 in e-ink display.

I have not tried out the pHAT since it is a smaller 2.13 in screen.

aceisace, 2022-04-29:
Hi there @CapnBio and welcome to Inkycal! Thanks for opening your first issue in this repo 💯 
Please note that the epd-displays from pimoroni are not officially supported by Inkycal (see [supported displays](https://github.com/aceisace/Inkycal/wiki#supported-e-paper-displays), hence they cannot be used with Inkycal (yet).
Technically it uses the same drivers as the waveshare 4.2" black-white-red or black-white-yellow display, but it will require some testing at least. Fortunately, I do have a black-white-red Inky what so it shouldn't take more than a few days. 

I'll leave this issue open until then. And yes, smaller displays don't serve good as full dashboards, hence bigger displays (4.2"<) are preferred for Inkycal. The largest supported display is the 12.48" available in black-white / black-white-colour :)

CapnBio, 2022-05-02:
Thanks lots for your support @aceisace I super appreciate it!

kengberg, 2022-11-17:
@aceisace so how were your trials?

aceisace, 2022-11-30:
Will get back to it this weekend 😅

pipplo, 2023-01-13:
I have an inkyWhat (yellow) I can try it out if you can point to where I can look to get started? (Otherwise I'll browse around and see if I can figure it out)

pipplo, 2023-01-15:
I tried a few different drivers and nothing happens on the inkyWhat for any of them. A quick (naive) browse of the inky project looks like their drivers are doing I2C as well as SPI, but otherwise I can't really compare the two driver parts well atm.

I was considering just importing inky (https://github.com/pimoroni/inky.git ) project and wrapping it with the driver calls as a new display type.

aceisace, 2023-01-15:
I've created a branch for this issue as the support for this display is unfortunately not as straight-forward as the pinout is different. I'm pretty sure the drivers are pretty similar with the 4.2" one from waveshare. Next weekend I will have some time to test this display with my pi4. I'll post an update after running some tests :+1:

closed 2023-03-23 by github-actions[bot]