aceinnolab/Inkycal #223: Can I help you with Brazilian Portuguese translation?

djairjr, 2022-11-27:
Please, how can I help translating InkyCal to Portuguese Brazil? What files can I move to do this job?

aceisace, 2022-11-30:
Hi, Inkycal does have built-in support for multiple languages, but not all are directly accessible. Which parts need translating?

djairjr, 2022-12-03:
I'm sorry, man. The Web interface (Web-UI) does not discriminate the Portuguese language. But I went through the configuration files and got the calendar in my language. I Web-UI section "Common module settings" does not have the Portuguese setting in language. To setup this language you just need to put pt on config file.

aceisace, 2023-01-13:
Hi again,
I've added the Portuguese language as an option in the web-ui:
[image]

Please check if it is working as expected 👍

aceisace, 2023-01-19:
Closing as issue seems stale. If you need to re-open this issue, please feel free to do so

closed 2023-01-19 by aceisace