aceinnolab/Inkycal #225: Bug in module inkycal_todoist

Semmidemmi263, 2022-12-29:
**Describe the bug**
Proposed todoist package in remark is wrong.

**To Reproduce**
There are several packages for the todoist_api. The installed one is wrong, the correct one should be: todoist_api_python.
Therefore, please change the prints following the importError exception to:
print('pip3 install todoist_api_python')

**Expected behavior**
Installation of the used package instead of the not used package "todoist-python", which, even if installed, still raises the ImportError. 
**Desktop (please complete the following information):**
 - OS: raspberry OS

**Smartphone (please complete the following information):**

**Additional context**
Add any other context about the problem here.

aceisace, 2023-01-08:
Same as #226 . Please let me know if the issue persists with that branch.

Semmidemmi263, 2023-01-10:
Hi aceisace, 
your change resolves the issue I had. Thank you!

closed 2023-01-10 by Semmidemmi263