aceinnolab/Inkycal #226: Bug in Todoist Module

Semmidemmi263, 2022-12-29:
**Describe the bug**
For several projects in the project_filter, the inkycal_todoist module only imports the first project. 

**To Reproduce**
Try to import more than one project from todoist.

**Expected behavior**
All tasks from all projects in the project_filter list should be imported and printed. Here, a small bug has been found: In the for-loop which is used to generate the image, the else case with break ("more todos than available lines") is lined up with "if todos:". It should line up with "if cursor < len(line_positions):". This change in the code has solved the issue for me.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: raspberry OS

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

aceisace, 2023-01-08:
Hi there @Semmidemmi263 . May I kindly ask you to check this branch of Inkycal:
https://github.com/aceisace/Inkycal/tree/feature/venv

If the issue persists there as well, please let me know. That branch is basically the new release containing several bugfixes. At the moment, I have tested it myself and it seems to work. To ensure that it really is working as expected, any help testing that branch is much appreciated so I can merge that branch into main and create a new release soon

Semmidemmi263, 2023-01-10:
Hi aceisace, 
The same goes for this issue: your fix helps to resolve it. Thank you!

closed 2023-01-10 by Semmidemmi263