aceinnolab/Inkycal #229: Updating display Error, 9.7 in

ChillippSFB, 2023-01-12:
Hi Aceisace,

i installed inkycal succesfully and also installed the 9.7in display driver and see the pikachu. But after executing the python inky_run.py i get the following error and nothing is showing on the display

todoist is not installed! Please install with:
pip3 install todoist-python
inkycal_jokes loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: 9_in_7
generating image(s) for iCanHazDad API - grab a random joke from icanhazdad api...OK!
Inkycal version: v2.0.0
Selected E-paper display: 9_in_7
Date: 13 Jan 23 | Time: 00:07
Generating images for all modules...successful
----------Started calibration of ePaper display----------
Calibrating... black... oops, something didn't work right :/
white...
oops, something didn't work right :/
Cycle 1 of 3 complete
Calibrating... black... oops, something didn't work right :/
white...
oops, something didn't work right :/
Cycle 2 of 3 complete
Calibrating... black... oops, something didn't work right :/
white...
oops, something didn't work right :/
Cycle 3 of 3 complete
-----------Calibration complete----------
Initialising..Updating display......oops, something didn't work right :/
Done
Sending E-Paper to deep sleep...Done

No errors since 1 display updates 
program started 22 seconds ago
53 minutes left until next refresh

aceisace, 2023-01-13:
Hi @ChillippSFB ,
please backup your `settings.json` file outside the Inkycal folder, then re-run the installation according to the new instructions. There was an issue with the todoist dependency and it should be resolved with the new release published just now. 
Do let me know if it works for you 👍

aceisace, 2023-01-18:
Did you manage to get it working again?

closed 2023-03-27 by github-actions[bot]