aceinnolab/Inkycal #232: [V2.02] Modules not installed via new installation process

zAm1987, 2023-01-13:
Hi,
as a long-time inkycal user I am very happy that a new version was officially released.

I directly installed it on a fresh Raspbian and followed the described (new) installation process. However running inkycal gave me the following PIL error:

"(venv) pi@raspberrypi:~/Inkycal $ python3 inky_run.py
Traceback (most recent call last):
  File "/home/pi/Inkycal/inky_run.py", line 2, in <module>
    from inkycal import Inkycal  # Import Inkycal
  File "/home/pi/Inkycal/inkycal/__init__.py", line 2, in <module>
    from inkycal.display import Display
  File "/home/pi/Inkycal/inkycal/display/__init__.py", line 1, in <module>
    from .display import Display
  File "/home/pi/Inkycal/inkycal/display/display.py", line 8, in <module>
    from PIL import Image
ModuleNotFoundError: No module named 'PIL'"

So i manually installed pillow using pip install. Then I tried it again and nearly the same error occured with another module. In the end i had to manually install a lot modules (todoist_api, recurring_ical_events, numpy feedparser, pyowm, arrow, ...). It's working fine now, but I think this is not the way it should be.  In addition the WebUI-URLS in the readme seems to be broken, too. So I just used my old file...

Maybe you can look into the installation process and the documented steps here on github (readme) if there is something missing?

greetings,
zAm1987

aceisace, 2023-01-13:
@zAm1987 Hi! 
Always happy to see long-term users of Inkycal! 😄
Thanks for reporting this issue. The tests on Github seem to be running fine in the installation process.

What happens when you run:
```bash
cd $HOME/Inkycal
source venv/bin/activate
python inky_run.py
```
(basically using `python` instead of `python3`)

Please ping me on Discord so I can add you to the inkycal-giveaway-members 👍

zAm1987, 2023-01-14:
@aceisace
Thanks for correcting my title to a one that fits better. :)

I startet with another fresh version of Raspberry PI OS (32-bit, "normal") and did the steps exactly as mentioned in the readme. During install (pip install -e ./) there was again an error about a missing rust compiler and therefore problems with cryptography. But I think this is not the problem. I went ahead and tried to run inkycal.

That still gave me the described PIL (and other modules) error. Running with python command instead of python3 command did not change anything.

Btw: The URLs to the WebUI are now working again.

greetings
zAm1987

pipplo, 2023-01-15:
> During install (pip install -e ./) there was again an error about a missing rust compiler and therefore problems with cryptography. But I think this is not the problem. I went ahead and tried to run inkycal.

I had the same issues with the rust compiler. I had to install rustc via apt. I also had an issue where inkycal wasn't running well if the wheels failed to install once. (i.e. i did the install -e, ran into errors, fixed them did install -e again and inkycal was bailing out.) After I had everything installing fine i deleted the entire venv and redid the install step and everything ran fine (except my inkyWhat doesn't work but that's a different issue :D)

obrocke, 2023-01-17:
Hi, I have the same problem. First I tried to upgrade my existing InkyCal version without success and now I started with a new installed Raspi Zero. I always get the same cryptography error message and if I ignore them, there are a lot of missing modules.
Is there any "easy" way to fix it?
Thanks for the project.

pipplo, 2023-01-17:
> Hi, I have the same problem. First I tried to upgrade my existing InkyCal version without success and now I started with a new installed Raspi Zero. I always get the same cryptography error message [...]

Can you paste the cryptography error?

Try 
`sudo apt install rustc`

And then delete the venv and rebuild from there.

aceisace, 2023-01-17:
Okay, looks like I'll add this instruction on the main branch for now. Don't have access to my Rpi yet, so leaving this issue open until I can test this on the Rpi.
@pipplo @obrocke If you are on Inkycal at Discord, please ping me for the Inkycal-giveaway badge 👍

closed 2023-01-17 by aceisace

obrocke, 2023-01-18:
I am not on Discord. Is there another way to receive the badge? Thanks

zAm1987, 2023-01-18:
Hi,

I just added rustc on the system and ran through the installation again. Somehow pip install -e ./ still puts out (now a lot more) errors during the cryptography part. Running inkycal after that puts out the described missing PIL and other errors.

Just adding rustc to the system does not seem to help. At least in my case.

aceisace, 2023-01-18:
@obrocke It's pretty simple to join Discord. And since the main discussions related to Inkycal happen there, you'll profit from better support. Since the giveaway will be done via a Discord Bot written by myself, it's a bit difficult to add someone without a discord account.

obrocke, 2023-01-19:
I have the same result as @zAm1987. Installed new after deleting venv and get the missing PIL message.