aceinnolab/Inkycal #233: Python version requirements

pipplo, 2023-01-14:
It's not super clear as far as I can see what the minimum python version requirement is. I had to find it by trial an error. The matplotlib dependency required 3.8, and after updating to 3.8 it looks like the todoist requirement in requirements.txt has a >= 3.9 requirement.

So through trial and error I've found that the minimum required (if you want todoist) is 3.9?

aceisace, 2023-01-15:
On my Mac I seem to have no issues running all tests with python3.8. As far as I remember, the Rpi's default python version of Raspberry Pi OS should be either python3.9 or python3.10, which should both be compatible with Inkycal.

Have you run into any issues so far during the installation of v2.0.2 or while running Inkycal or the tests?

pipplo, 2023-01-15:
Interesting. I was just running on the OS i already had installed (astroberry) which is on an older version of raspbian OS that only had python 3.7 installed.  So on that system matplotlib required 3.8, then todoist was specifying a 3.9 requirement. But then after installing 3.9 another package started requiring libffi7 and i only had libffi6. So in the end I just got a new sd card and flashed the latest raspbian os onto it.

Everything seems to be working fine now since it defaulted to 3.9. The only thing I had to do was apt install rustc (which maybe should be added to the initial dependencies?)

aceisace, 2023-01-16:
Glad to hear it's working as expected. Let's see, rustc? I don't remember this as a requirement. At which step does this cause a problem?

pipplo, 2023-01-16:
When i'm running the pip install, it's a requirement of building the cryptography pip module.

aceisace, 2023-01-17:
This issue seems related to #232 so closing this issue to prevent duplicates.
I've also changed the minimum python requirement from python3.7 to python3.8 since that is the one I can confirm works well so far. This is now part of the badges at the top of the readme

closed 2023-01-17 by aceisace