aceinnolab/Inkycal #235: Feature request: Adjustable font size (on agenda)

zAm1987, 2023-01-18:
Hi,

in my usage-scenario the typical viewing distance from my family and the inkycal-display is about 1.5 meters. I am using the 7.5inch v1 display and sometimes it is hard to read the entries on the agenda. Therefore in an older version (long time ago) I made the fontsize a bit bigger via a value in one of the .py files. That lowered the distortion in the rendering, too.

In the current version I can't find a (hidden) setting for the filesize anymore. Could such a setting officially be implemented?

I guess that would enable the users to better adjust inkycal to their needs.

greetings,
zAm1987

aceisace, 2023-01-18:
Hi,
[image]

Does this setting not work on the agenda module?

zAm1987, 2023-01-19:
Ah there it is! After the last update the font-size changed to default and I searched everywhere but the settings file. Thought already I only dreamed of changing the size in the past. In the end my (reused) settings file was somehow broken.

Anyway, I created a brand new settings file via WebUI, rebooted and the font-size changed back to my usual value.

Sorry for my unnecessary request Here! 🙃😅

aceisace, 2023-01-19:
Haha, no worries, the changes weren't that obvious :)
Perfect! Glad to hear this issue was resolved. Closing 👍

closed 2023-01-19 by aceisace