aceinnolab/Inkycal #236: Add border around modules

ZardoZzZz, 2023-01-18:
To make modules more defined it would be nice to have an option to add a thin border around each module. 
I have a iCalendar module and below this is an RSS module but it's difficult to know where one module ends and the other begins.

aceisace, 2023-01-22:
How does this look?
[image]

ZardoZzZz, 2023-01-23:
Yep looks great thanks , maybe a 1 character padding between the border and the first character ?

aceisace, 2023-11-24:
Implemented border feature in this [commit](https://github.com/aceinnolab/Inkycal/commit/60be01fbdb948ad9bca63d4d495c651c0bb860d3)

Closing this issue as the border feature has been implemented and the unittests are passing.

closed 2023-11-24 by aceisace

ZardoZzZz, 2023-11-24:
Hi Ace,

I see this is closed now, what is a border feature ?
And is it working now ,  what do I need to do on my RP to get the weather to my local time zone?

Thanks for all the help have a great weekend.

EXTERNAL EMAIL: This email originated from outside of our organization. DO NOT CLICK links, open attachments, or respond to this email unless you recognize the sender and know the content is safe.

aceisace, 2023-11-24:
Hi there,
the border feature basically draws a 1px border around each module when the settings in the settings.json file is set to show the border. The main aim is a better fine-tuning of each module's height so that the already limited space on the display can be utilised more efficiently.

You can try out the feature with the following steps:
```sh
sudo apt-get install wkhtmltopdf -y
cd ~/Inkycal
git pull
source venv/bin/activate
pip install -r requirements.txt
# To test if Inkycal is still running fine:
python inky_run.py
```

If the test works fine, you can reboot the Raspberry Pi (`sudo reboot`) so the changes are applied after the reboot.

Concerning the timezone feature, this is currently in progress (see the other issue) and will be finished soon. One of the main reasons this got delayed quite a bit was due to the pyowm library, which is not being maintained anymore since years and would eventually lead to larger problems in the long-term. As a result, a custom wrapper had be implemented to achieve the same, but without the pyowm library. The tests so far are looking good, so more weather-related features can be implemented, including fixing the timezone.

ZardoZzZz, 2023-11-24:
Thanks Ace 🙂

aceisace, 2023-11-24:
You're welcome!