aceinnolab/Inkycal #239: Improve efficiency

lukasMega, 2023-01-23:
👋 First of all, thank you for this project! Very nice and useful idea! :) 👍

## Description
I'm using Raspberry Pi Zero W (1-core CPU) and inkycal python scripts takes ~2 min (I see 100% CPU usage approx. ~2 min)

**To Reproduce**
I will update details later.

## My configuration / environment

- Inkycal was installed: August 2022

- HW: [Raspberry Pi Zero W](https://www.raspberrypi.com/products/raspberry-pi-zero-w/)
  - SoC: Broadcom BCM2835 | 1 GHz single-core ARM1176JZF-S
  - RAM: 512 MB
- Memory card: [I will update later]

- OS: minimal with no GUI - DietPi (debian 11, kernel 5.15.…)
- installed additional packages: only those required for inkycal + htop + avahi-daemon
- RAM usage: ~23 — 80 MB (with running inky_run.py)

- ePaper: [Waveshare 7.5"](https://www.waveshare.com/7.5inch-e-paper-hat.htm) (800x480) most probably rev. v2

------------------------

ℹ️ I know, it's old and slow hardware, maybe it's nothing to improve. I will try to put here some more useful information from my investigation.

------------------------

## 🧾 output of some commands:
```bash
# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Running services: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dietpi@rpi0Paper:~$ sudo service --status-all
 [ + ]  avahi-daemon
 [ - ]  console-setup.sh
 [ + ]  cron
 [ + ]  dbus
 [ + ]  dropbear
 [ + ]  fake-hwclock
 [ - ]  hwclock.sh
 [ - ]  keyboard-setup.sh
 [ + ]  kmod
 [ + ]  networking
 [ + ]  procps
 [ + ]  rngd
 [ - ]  rsync
 [ - ]  sudo
 [ + ]  udev

dietpi@rpi0Paper:~$ sudo dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status 

[  OK  ] DietPi-Services | avahi-daemon		 active (running) since Sat 2023-01-21 17:38:54 CET; 1 day 18h ago
[  OK  ] DietPi-Services | cron			 active (running) since Sat 2023-01-21 17:38:54 CET; 1 day 18h ago
[  OK  ] DietPi-Services | dropbear		 active (running) since Sat 2023-01-21 17:38:47 CET; 1 day 18h ago
[ INFO ] DietPi-Services | dietpi-vpn		 inactive (dead)
[ INFO ] DietPi-Services | dietpi-cloudshell	 inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog	 active (exited) since Sat 2023-01-21 17:38:47 CET; 1 day 18h ago
[  OK  ] DietPi-Services | dietpi-preboot	 active (exited) since Sat 2023-01-21 17:38:48 CET; 1 day 18h ago
[  OK  ] DietPi-Services | dietpi-postboot	 active (exited) since Sat 2023-01-21 17:38:53 CET; 1 day 18h ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor	 inactive (dead)

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ disk usage: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dietpi@rpi0Paper:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        15G  5.7G  8.4G  41% /
devtmpfs        207M     0  207M   0% /dev
tmpfs           239M     0  239M   0% /dev/shm
tmpfs            96M  1.7M   94M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           1.0G     0  1.0G   0% /tmp
tmpfs            50M  8.0K   50M   1% /var/log
/dev/mmcblk0p1  127M   53M   74M  42% /boot
dietpi@rpi0Paper:~$ 

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ number of installed packages: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dietpi@rpi0Paper:~$ sudo dpkg-query -l | cat | wc --lines
307
dietpi@rpi0Paper:~$ dpkg --list | wc --lines
307
dietpi@rpi0Paper:~$ dpkg -l | grep -c '^ii'
302

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ htop: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CPU[||||                                                                                  3.2%] Mem[|||||||||||||||||||||||||||||||||||||||||||                                     86.3M/477M]
Avg[||||                                                                                  3.2%] Swp[                                                                                  0K/1.53G]
                                                                                                Tasks: 18; 1 running
                                                                                                Load average: 0.55 0.94 0.59 
                                                                                                Uptime: 00:09:19
  PID△USER        RES  NI CPU%   TIME+  Command
    1 root       8496   0  0.0  0:07.07 /sbin/init
   98 root       5944   0  0.0  0:00.90 ├─ /lib/systemd/systemd-journald
  122 root       3996   0  0.0  0:00.72 ├─ /lib/systemd/systemd-udevd
  180 messagebu  3484   0  0.0  0:00.14 ├─ /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
  189 root       1284   0  0.0  0:00.02 ├─ /usr/sbin/rngd -f
  216 root       1452   0  0.0  0:00.00 ├─ /usr/sbin/dropbear -p 22 -W 65536
  494 root       2472   0  0.0  0:02.06 │  └─ /usr/sbin/dropbear -p 22 -W 65536
  600 dietpi     4480   0  0.0  0:00.76 │     └─ -bash
  797 root       4040   0  0.0  0:00.08 │        └─ sudo htop
  798 root       3776   0  3.2  0:00.88 │           └─ htop
  319 root       3992   0  0.0  0:00.04 ├─ /sbin/wpa_supplicant -s -B -P /run/wpa_supplicant.wlan0.pid -i wlan0 -D nl80211,wext -c /etc/wpa_supplicant/wpa_supplicant.conf
  325 root       5068   0  0.0  0:00.07 ├─ /sbin/dhclient -4 -v -i -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases -I -df /var/lib/dhcp/dhclient6.wlan0.leases wlan0
  375 root       1924   0  0.0  0:00.02 ├─ /sbin/agetty -o -p -- \u --noclear tty1 linux
  400 avahi      3044   0  0.0  0:00.13 ├─ avahi-daemon: registering [rpi0Paper.local]
  401 avahi      1260   0  0.0  0:00.00 │  └─ avahi-daemon: chroot helper
  404 root       2388   0  0.0  0:00.03 ├─ /usr/sbin/cron -f
  420 root       4000   0  0.0  0:00.09 └─ sudo python3 /home/dietpi/inky_run.py
  661 root      86300   0  0.0  4:39.17    └─ python3 /home/dietpi/inky_run.py

F1Help  F2Setup F3SearchF4FilterF5List  F6SortByF7Nice -F8Nice +F9Kill  F10Quit

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Dietpi benchmark: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#  sudo dietpi-config > 10. Tools > Benchmarks > DietPi-Benchmark

Benchmarks completed:                                           
 - CPU performance : Duration = 307.34 seconds (lower is faster)
 - CPU temps       : Idle = 32 °C | Full load = 36 °C           
 - RootFS I/O      : Write = 10 MiB/s | Read = 17 MiB/s         
 - RAM I/O         : Write = 129 MiB/s | Read = 210 MiB/s       

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  DietPi CPU info: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dietpi@rpi0Paper:~$ sudo cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv6l
 Temperature  |     32 °C / 89 °F : Cool runnings
 Governor     |     ondemand
 Throttle up  |     85% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      800 MHz      700 MHz    800 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.

# ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ Network speed test: ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
dietpi@rpi0Paper:~$ speedtest-cli --simple
Ping: 93.515 ms
Download: 24.64 Mbit/s
Upload: 23.09 Mbit/s

```

aceisace, 2023-01-23:
@lukasMega Thanks for opening your first issue on Inkycal 💯 . Glad to hear you like Inkycal 😄 

Thanks for all the profiling! Indeed I did not know about these, but I had a feeling that there still is lot which can be improved. While generating the images, the load is dependent on several factors, including the selected modules and e.g. size of the calendar file.

Currently, the script doesn't "wait" while running non-stop but instead invokes the time.sleep() function of python, which still stresses the cpu. A better approach would be to use timed threads instead of sleep, but this is a bit complicated and I only recently learnt about it. 

If you have suggestions in regards to reducing the cpu stress by e.g. changes in code, please do share them. Any help with reducing the cpu stress is most welcome.

lukasMega, 2023-01-25:
I don't think that using sleep() function is wrong, also it don't uses CPU time.

I did small update: I disabled (commented) `inky.test()` in inky_run.py: 

```c#
~$ date +"%T.%N"; time sudo python3 /home/dietpi/inky_run.py ; date +"%T.%N";
10:27:24.773842503
inkycal_calendar loaded
inkycal_weather loaded
inkycal_feeds loaded
loaded inkycal
Inkycal version: v2.0.0
Selected E-paper display: epd_7_in_5_v2
Date: 25 Jan 23 | Time: 10:27
Generating images for all modules...successful
Initialising..Updating display......Done
Sending E-Paper to deep sleep...Done

No errors since 1 display updates 
program started 2 minutes ago

real	1m59.310s
user	1m45.307s
sys	0m6.012s
10:29:24.113349503
```

and takes not 3 min 15 sec but 1 min 59 sec :)

aceisace, 2023-01-25:
`inky.test()` only tries to generate all images once before the actual run. This is useful if you e.g. changed your config and Inkycal doesn't work anymore. So the test is actually not required. Depending on the display's size and supported colours, the time may increase or decrease too.

Looking forward to your enhancements when they are ready. When they are, please create a PR pointing to the development branch. Closing this issue 👍

closed 2023-01-25 by aceisace