aceinnolab/Inkycal #240: Cron to auto-launch Inky-run after restart is no longer working on 2.0.2

AndBosta, 2023-01-23:
**TL;DR - Crontab is not auto launching InkyCal on boot on a RasPi.**

I recently updated my Inky Cal to 2.0.2... 
Everything is mostly back to correct, but the steps to setup the cron found on "Running on each boot" are no longer working for me (it was good on one of the previous versions of InkyCal).

Attached is what I see when I SSH into the RasPi and run 'crontab -e'
[image]

If I SSH into the RasPi after startup, I don't see the correct process when running 'htop' (process viewer).

From the terminal, I can manually start InkyCal with the following commands, and it updates the display and runs as expected:
cd $HOME/Inkycal
source venv/bin/activate
python3 inky_run.py

Thanks for looking into this!

zAm1987, 2023-01-23:
Today I realized the same after a reboot. Crontab solution is not working in my case, too. I think had not noticed this before because I started inkycal manually a lot the last days while tinkering with it.

aceisace, 2023-01-23:
@AndBosta @zAm1987 Thanks for your feedback and this info. Indeed, switching to virtual environment did break the crontab which previously used to work. I've found a fix for this, which basically boils down to running the following:
```bash
# reset crontab
crontab -r
# add updated command
(crontab -l ; echo "@reboot sleep 60 && cd $HOME/Inkycal && venv/bin/python inky_run.py &")| crontab -
```
Please let me know if this works for you as well. Then this issue can be closed 👍

AndBosta, 2023-01-23:
@aceisace - Yes, this looks to have worked. Display has updated on boot and at the top of the hour after I implemented the change. Thank you!

closed 2023-01-23 by AndBosta