aceinnolab/Inkycal #241: Daily weather is not respecting time zones and displays tommorows weather 8 hours too early.

ZardoZzZz, 2023-01-23:
The issue I have is the daily weather is not showing the weather for today but its showing the weather for tommorow onwards (in the morning it's showing the correct weather for today but at about 2pm it changes the weather for tommorows weather.

aceisace, 2023-11-08:
Apologies for getting back so late, I'm working on this issue now, could you provide the timezone for your place? Then I can reproduce this error 👍

ZardoZzZz, 2023-11-08:
Hi Ace,

Thanks for looking into this .. my time zone is Vancouver(Canada) so it’s Pacific standard time (-8 GMT)

EXTERNAL EMAIL: This email originated from outside of our organization. DO NOT CLICK links, open attachments, or respond to this email unless you recognize the sender and know the content is safe.

closed 2024-02-08 by aceisace