aceinnolab/Inkycal #249: v2.0.2 not working: Illegal instruction

weidingerc, 2023-04-02:
Hi all,

I wanted to update to v2.0.2 from v2.0.0.

To have a clean install if started completely new by installing the newest Raspbian OS. 
I then followed the steps described in the readme. 
Unfortunately when starting inky_run.py it always returns "Illegal instruction".

What am I doing wrong? :/

aceisace, 2023-04-03:
Hi there,
thanks for reporting this issue. Could you please provide the full traceback or a a screenshot of the error?

This issue is a bit more difficult to fix since it's basically an incompatibility between the OS and a package. In the meantime, you can try using InkcalOS from the releases page (v2.0.1)

weidingerc, 2023-04-03:
Here's the screenshot

[image]

Ok, I can try 2.0.1 in the meantime.

aceisace, 2023-04-05:
Thanks for the screenshot! I'll fix this as soon as I can. It seems this is somehow very specific to the current release of Raspberry Pi OS and python. On my Mac, I cannot reproduce it, but I'll test this on my Raspberry Pi soon.

Just for reference, could you provide some additional details about the setup? Which version of Raspberry Pi python are you using? To find the python version within the virtual environment, you can type `which python` or `which python3`

weidingerc, 2023-04-06:
Ok I guess I found the issue. When I type python -v or python3 -v the following output occurs:

```
(venv) pi@raspberrypi:~/Inkycal $ python -v
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
# installing zipimport hook
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
import 'zipimport' # <class '_frozen_importlib.FrozenImporter'>
# installed zipimport hook
# /usr/lib/python3.9/encodings/__pycache__/__init__.cpython-39.pyc matches /usr/lib/python3.9/encodings/__init__.py
# code object from '/usr/lib/python3.9/encodings/__pycache__/__init__.cpython-39.pyc'
# /usr/lib/python3.9/__pycache__/codecs.cpython-39.pyc matches /usr/lib/python3.9/codecs.py
# code object from '/usr/lib/python3.9/__pycache__/codecs.cpython-39.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0xb678ba90>
# /usr/lib/python3.9/encodings/__pycache__/aliases.cpython-39.pyc matches /usr/lib/python3.9/encodings/aliases.py
# code object from '/usr/lib/python3.9/encodings/__pycache__/aliases.cpython-39.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0xb6727a30>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0xb678b988>
# /usr/lib/python3.9/encodings/__pycache__/utf_8.cpython-39.pyc matches /usr/lib/python3.9/encodings/utf_8.py
# code object from '/usr/lib/python3.9/encodings/__pycache__/utf_8.cpython-39.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0xb678b8e0>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.9/encodings/__pycache__/latin_1.cpython-39.pyc matches /usr/lib/python3.9/encodings/latin_1.py
# code object from '/usr/lib/python3.9/encodings/__pycache__/latin_1.cpython-39.pyc'
import 'encodings.latin_1' # <_frozen_importlib_external.SourceFileLoader object at 0xb6727b38>
# /usr/lib/python3.9/__pycache__/io.cpython-39.pyc matches /usr/lib/python3.9/io.py
# code object from '/usr/lib/python3.9/__pycache__/io.cpython-39.pyc'
# /usr/lib/python3.9/__pycache__/abc.cpython-39.pyc matches /usr/lib/python3.9/abc.py
# code object from '/usr/lib/python3.9/__pycache__/abc.cpython-39.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0xb6727e80>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0xb6727d00>
# /usr/lib/python3.9/__pycache__/site.cpython-39.pyc matches /usr/lib/python3.9/site.py
# code object from '/usr/lib/python3.9/__pycache__/site.cpython-39.pyc'
# /usr/lib/python3.9/__pycache__/os.cpython-39.pyc matches /usr/lib/python3.9/os.py
# code object from '/usr/lib/python3.9/__pycache__/os.cpython-39.pyc'
# /usr/lib/python3.9/__pycache__/stat.cpython-39.pyc matches /usr/lib/python3.9/stat.py
# code object from '/usr/lib/python3.9/__pycache__/stat.cpython-39.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0xb673cf40>
# /usr/lib/python3.9/__pycache__/_collections_abc.cpython-39.pyc matches /usr/lib/python3.9/_collections_abc.py
# code object from '/usr/lib/python3.9/__pycache__/_collections_abc.cpython-39.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0xb673cf58>
# /usr/lib/python3.9/__pycache__/posixpath.cpython-39.pyc matches /usr/lib/python3.9/posixpath.py
# code object from '/usr/lib/python3.9/__pycache__/posixpath.cpython-39.pyc'
# /usr/lib/python3.9/__pycache__/genericpath.cpython-39.pyc matches /usr/lib/python3.9/genericpath.py
# code object from '/usr/lib/python3.9/__pycache__/genericpath.cpython-39.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0xb66e67c0>
import 'posixpath' # <_frozen_importlib_external.SourceFileLoader object at 0xb66d2448>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0xb673c628>
# /usr/lib/python3.9/__pycache__/_sitebuiltins.cpython-39.pyc matches /usr/lib/python3.9/_sitebuiltins.py
# code object from '/usr/lib/python3.9/__pycache__/_sitebuiltins.cpython-39.pyc'
import '_sitebuiltins' # <_frozen_importlib_external.SourceFileLoader object at 0xb673cd00>
# /usr/lib/python3.9/__pycache__/_bootlocale.cpython-39.pyc matches /usr/lib/python3.9/_bootlocale.py
# code object from '/usr/lib/python3.9/__pycache__/_bootlocale.cpython-39.pyc'
import '_locale' # <class '_frozen_importlib.BuiltinImporter'>
import '_bootlocale' # <_frozen_importlib_external.SourceFileLoader object at 0xb66e6a78>
# /home/pi/Inkycal/venv/lib/python3.9/site-packages/_distutils_hack/__pycache__/__init__.cpython-39.pyc matches /home/pi/Inkycal/venv/lib/python3.9/site-packages/_distutils_hack/__init__.py
# code object from '/home/pi/Inkycal/venv/lib/python3.9/site-packages/_distutils_hack/__pycache__/__init__.cpython-39.pyc'
import '_distutils_hack' # <_frozen_importlib_external.SourceFileLoader object at 0xb66e6e08>
[... cut ...]
# extension module 'readline' executed from '/usr/lib/python3.9/lib-dynload/readline.cpython-39-arm-linux-gnueabihf.so'
import 'readline' # <_frozen_importlib_external.ExtensionFileLoader object at 0xb66e6c88>
import 'atexit' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.9/__pycache__/rlcompleter.cpython-39.pyc matches /usr/lib/python3.9/rlcompleter.py
# code object from '/usr/lib/python3.9/__pycache__/rlcompleter.cpython-39.pyc'
import 'rlcompleter' # <_frozen_importlib_external.SourceFileLoader object at 0xb66ff2c8>
```
>>>

tinolinoh, 2023-04-07:
Hi there,

first of all, thank you for your amazing work! 
Unfortunately I am experiencing exactly the same issue:

[image]

Regards,

Tino

lukasMega, 2023-04-12:
The same here with 2.0.2 on Raspberry Pi Zero W (ARM v6) 
`Linux RPi0paper 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux`

lukasMega, 2023-04-12:
I found a solution (most probably). At least I managed to running on my setup:
```
device:  Raspberry Pi Zero W (ARM v6)
OS:      DietPi
kernel:  Linux RPi0paper 6.1.21+ #1642 Mon Apr  3 17:19:14 BST 2023 armv6l GNU/Linux
Inkycal: 2.0.2
```

if command:
`python3 -c "import yfinance"` outputs: `Illegal instruction` then: 
 - comment out **stocks** related code in: `inkycal/init.py` and `inkycal/modules/init.py`

try running inkycal...

for `ImportError: The _imagingft C module is not installed`:
 - install: `sudo apt install libfreetype6-dev`

for `PermissionError` (`PermissionError: [Errno 13] Permission denied`):
 and also `RuntimeError: No access to /dev/mem`:

 - I'm not sure what helped, but I executed:
 -- `sudo chmod 666 /dev/spidev0.0`
-- `sudo adduser dietpi gpio`
-- `sudo usermod -a -G gpio dietpi`
 - note that my user is not `pi` but `dietpi`

---

and finally log out and log in (or restart). DONE 🏁

aceisace, 2023-04-16:
I've added the suggested modifications here #251 . @lukasMega  Could you take a look at the changes to ensure they are correct before merging?

aceisace, 2023-05-10:
Closing this issue as the problem is resolved and the actions are working again. If you feel a need to re-open this issue, please do so.

closed 2023-05-10 by aceisace