aceinnolab/Inkycal #260: Compilation error during installation

pokapple, 2023-06-24:
Whenever I try to install Inkycal on my Raspberry Pi Zero, at some point, the installation stucks and nothing happens anymore.
I did the installation on a freshly installed Raspberry Pi OS and did every step as reported in the installation manual.

The installation stucks at this point:
 Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... \

To get more information, I ran the code with:
 pip install --verbose -e ./

The code terminates at this point:

Installing collected packages: patchelf
  Successfully installed patchelf-0.17.2.1
  Installing backend dependencies ... done
  Running command Preparing metadata (pyproject.toml)
  + meson setup /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657 /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657/.mesonpy-p3lqrr0k/build -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md --vsenv --native-file=/tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657/.mesonpy-p3lqrr0k/build/meson-python-native-file.ini
  The Meson build system
  Version: 1.1.1
  Source dir: /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657
  Build dir: /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657/.mesonpy-p3lqrr0k/build
  Build type: native build
  Project name: contourpy
  Project version: 1.1.0
  C++ compiler for the host machine: c++ (gcc 10.2.1 "c++ (Raspbian 10.2.1-6+rpi1) 10.2.1 20210110")
  C++ linker for the host machine: c++ ld.bfd 2.35.2
  Host machine cpu family: arm
  Host machine cpu: armv6l
  Program python found: YES (/home/pi/Inkycal/venv/bin/python)
  Found pkg-config: /usr/bin/pkg-config (0.29.2)
  pybind11-config found: YES (/tmp/pip-build-env-5mz_dx3g/overlay/bin/pybind11-config) 2.10.4
  Run-time dependency pybind11 found: YES 2.10.4
  Configuring _build_config.py using configuration
  Build targets in project: 1

  contourpy 1.1.0

    Build config
      host_cpu    : armv6l
      build_cpu   : armv6l
      cross_build : False
      optimization: 3

    User defined options
      Native files: /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657/.mesonpy-p3lqrr0k/build/meson-python-native-file.ini
      buildtype   : release
      vsenv       : True
      b_ndebug    : if-release
      b_vscrt     : md

  Found ninja-1.11.1.git.kitware.jobserver-1 at /tmp/pip-build-env-5mz_dx3g/overlay/bin/ninja

  Visual Studio environment is needed to run Ninja. It is recommended to use Meson wrapper:
  /tmp/pip-build-env-5mz_dx3g/overlay/bin/meson compile -C .
  + /tmp/pip-build-env-5mz_dx3g/overlay/bin/ninja -v
  [1/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/chunk_local.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/chunk_local.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/chunk_local.cpp.o -c ../../src/chunk_local.cpp
  [2/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/fill_type.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/fill_type.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/fill_type.cpp.o -c ../../src/fill_type.cpp
  [3/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/line_type.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/line_type.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/line_type.cpp.o -c ../../src/line_type.cpp
  [4/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/converter.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/converter.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/converter.cpp.o -c ../../src/converter.cpp
  [5/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005.cpp.o -c ../../src/mpl2005.cpp
  [6/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005_original.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005_original.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2005_original.cpp.o -c ../../src/mpl2005_original.cpp
  [7/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/outer_or_hole.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/outer_or_hole.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/outer_or_hole.cpp.o -c ../../src/outer_or_hole.cpp
  [8/14] c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o -c ../../src/mpl2014.cpp
  FAILED: src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o
  c++ -Isrc/_contourpy.cpython-39-arm-linux-gnueabihf.so.p -Isrc -I../../src -I/usr/include/python3.9 -I/tmp/pip-build-env-5mz_dx3g/overlay/lib/python3.9/site-packages/pybind11/include -I/usr/include/arm-linux-gnueabihf/python3.9 -fvisibility=hidden -fdiagnostics-color=always -DNDEBUG -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Werror -std=c++17 -O3 -fPIC -DCONTOURPY_VERSION=1.1.0 -MD -MQ src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o -MF src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o.d -o src/_contourpy.cpython-39-arm-linux-gnueabihf.so.p/mpl2014.cpp.o -c ../../src/mpl2014.cpp
[... cut ...]
  full command: /home/pi/Inkycal/venv/bin/python /home/pi/Inkycal/venv/lib/python3.9/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpnj0msrrz
  cwd: /tmp/pip-install-refl93sh/contourpy_12318d7b8cd4443bbe2bc8c68971d657
  Preparing metadata (pyproject.toml) ... error
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Any ideas?

aceisace, 2023-06-24:
Hi @pokapple and thanks for informing about the issue with the very long installation time.
I did some investigation and found out that this issue is caused by the cryptography dependency used by Inkycal. I ran the installation on my Raspberry Pi 4 just now and it seems that although it takes a very long time (20 mins on the Rpi4), it does complete without errors. However, this is hardly a solution for most users as they are using the Zero W.

I will need your help to fix this issue by testing the following:
1) navigate to the Inkycal folder:
```sh
cd $HOME
cd Inkycal
```

2) rename the current `requirements.txt` file, e.g.
 ```sh
mv requirements.txt requirements-old.txt 
```

3) create a new `requirements.txt` file with the following content:
```bash
nano requirements.txt
```
add the following content:
```py
arrow==1.2.3
certifi==2023.5.7
cycler==0.11.0
feedparser==6.0.10
fonttools==4.40.0
geojson==2.3.0
icalendar==5.0.7
kiwisolver==1.4.4
lxml==4.9.2
matplotlib==3.7.1
multitasking==0.0.11
numpy==1.25.0
packaging==23.1
pandas==2.0.2
Pillow==9.5.0
pyowm==3.3.0
pyparsing==3.1.0
PySocks==1.7.1
python-dateutil==2.8.2
pytz==2023.3
recurring-ical-events==2.0.2
requests==2.31.0
sgmllib3k==1.0.0
six==1.16.0
todoist-api-python==2.0.2
typing_extensions==4.6.3
urllib3==2.0.3
yfinance==0.2.21
python-dotenv==1.0.0
setuptools==68.0.0
```

then press on the [CTRL] + [X] key of your keyboard, confirm with [Y] and press [ENTER].

4) Attempt to re-install inkycal:
```bash
source venv/bin/activate
pip install -e ./
```

Please let me know once you have attempted this. Ideally, this fill make the installation much faster than with the cryptography module.

pokapple, 2023-06-24:
Thank you very much for your immediate answer!
I've tried your solution and it really works fine! The installation finishes and Inkycal is also running.

I found another workaround for the original installation. When the installation crashed, the Pi started blinking continously, indicating a memory problem.
I changed the size of the swap file from 100 MB to 1 GB by using:

```sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile
CONF_SWAPSIZE=1024
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
sudo reboot```

It took a lot of time, but with additional memory, the installation finished smoothly.

aceisace, 2023-06-24:
Thanks for letting me know @pokapple . I've adapted the requirements.txt file so it does not attempt to install the cryptography package anymore, hence speeding up the installation progress. 

Thank you for sharing the workaround for the memory issue on the Zero W! I'll make a note of these 👍 

I'll be closing this issue now as the problem has been resolved. If you feel a need to re-open this issue, you may do so

closed 2023-06-24 by aceisace