aceinnolab/Inkycal #262: Installation freezes while installing matplotlib

DocSchnabel, 2023-07-21:
After typing: Install `pip -e /.` it gets stuck while trying to install matplotlib. It freezes for at least 5 hours. Even installing matplotlib manually results in the same error.

[image]

I´m trying to install inkycal on a **Raspberry Pi Zero WH** and I have installed **PiOS** from Raspberry Pi Imager.

aceisace, 2023-07-22:
@DocSchnabel Thanks for opening this issue from Discord. 
It seems this is a common issue on the Raspberry Pi Zero W, which may be caused by running out of memory. As a possible fix, please try the following:
```sh
cd ~/Inkycal && source venv/bin/activate
pip install --no-cache-dir numpy
pip install --no-cache-dir matplotlib
```

aceisace, 2023-07-29:
Filling in some details from the discussions on Discord. It seems this issue is not related to Inkycal itself, but rather the limited memory on the Raspberry Pi Zero W. As a result, a large installation can sometimes freeze. A temporary fix would be to increase the swapsize like this:

```sh
sudo dphys-swapfile swapoff
sudo nano /etc/dphys-swapfile
# change the CONF_SWAPSIZE  line so it looks like this: CONF_SWAPSIZE=1024
sudo dphys-swapfile setup
sudo dphys-swapfile swapon
```

This issue seems a little random, as not all users with the same setup in the Zero are experiencing this issue. As a result inkycal-os-lite is now available for sponsors, whose contribution is used to cover ongoing expenses of Inkycal. 
Inkycal OS Lite was built to be minimal and hence does not have a Web-UI as this would strain the cpu of the Zero even further. But it is fully compatible with the Web-UI from the README. With Inkycal pre-installed, spi enabled and auto-start on boot enabled, it allows a zero-code version of Inkycal for the end-user.

Closing this issue as some discord users have confirmed that the swapfile does indeed fix the memory issue. Should you feel a need to re-open this issue, please feel free to do so.

closed 2023-07-29 by aceisace