aceinnolab/Inkycal #271: ModuleNotFoundError: No module named 'PIL'

Ryan5five5, 2023-09-25:
Hi,
I have done a manual install of 2.0.2 following the instructions line by line.
When I run Inkycal it has the following errors.  I do have the settings.json file in the /boot directory.
[image]

aceisace, 2023-09-25:
Hi @Ryan5five5 , 
thank you very much for your interest in Inkycal! The issue in this case is the activation of the virtual environment. 
You can image a python virtual environment as a container used only by Inkycal to install libraries and versions which may be different from the version currently installed on your system. Therefore, you can cleanly install and uninstall Inkycal with virtual environments. The only catch is that virtual environments have to be activated and deactivated when working with them, like this:
```bash
cd Inkycal
# command to activate Inkycal
source venv/bin/activate
# now you  can do stuff inside the virtual environment:
python inky_run.py
# deactivate the virtual environment
deactivate
```

If the same error still persists, you may need to (re-) install the required libraries, like this:
```sh
cd Inkycal
source venv/bin/activate
pip install -r requirements.txt
```

Ryan5five5, 2023-09-26:
Thank you for the help, the screen shot did not show it but I did also do the 
cd Inkycal 
source venv/bin/activate
commands beforehand.
I tried pip install -r requirements.txt 
it did load a bunch of files but it still did not run (same problem)
It did say the hashes did not match but I don't think that would matter?

aceisace, 2023-09-26:
I tried to reply via mail for your other query, but the delivery failed again.
In regards to the query via mail, which version of Raspberry Pi are you using? 3,4,0?

In regards to this query, please attach the log when running the pip install command in the activated virtual environment. The tests are all running correctly on Raspberry Pi, so it is most likely an issue with the installation. An error message would be great for debugging

aceisace, 2023-09-26:
Also, the kernel panic issue is related to the medium used to flash the OS, but it's difficult to pinpoint where exactly something went wrong. I'll re-generate the latest version again and will provide you with a link

Ryan5five5, 2023-09-28:
I have formatted the SD card and manually reinstalled.  Comes up the same error as above and also the same error after using the pip install -r requirements.txt command.  Its running on the Pi Zero.  How do I get a log file?

PS I'm still getting Kernel panic using the img file.

aceisace, 2023-10-01:
Apologies for the late reply as I have been a little too busy the past days @Ryan5five5 
I have tested the installation from scratch of Inkycal OS Lite once again on my zero w and it seems to work just fine with Raspberry Pi Imager when flashing on an empty microSD card. I've re-sent the Inkycal OS Lite file to you via mail. Please give this a try and let me know if it works

Ryan5five5, 2023-10-15:
Not sure if you got my email but its all working fine now thank you.

aceisace, 2023-10-15:
Apologies, it seems the mail did not reach me for some reason, but thanks for letting me know! Glad to hear you got it working now and hope you'll enjoy Inkycal 😄 . Closing this issue now

closed 2023-10-15 by aceisace