aceinnolab/Inkycal #280: Problem with the weather module.

jamesvcc, 2023-11-10:
No matter where I put the weather module on the screen, it does not render the whole border correctly.  The top part of the border is missing.

I have picture that describes what I am seeing exactly - but I have no idea how to attach it to this issue.

Thanks,
James

jamesvcc, 2023-11-10:
Ah ha, worked out how to attach an image.

[image]

aceisace, 2023-11-10:
Hi James, thanks for reporting this issue!
It's not clear yet whether or not this is a bug or just a configuration problem, but assuming it might be an issue with the paddings, you can try this:
```
sudo nano /boot/settings.json
# find the config section of the weather module,
# change these two values to 10:
"padding_x": 10,
"padding_y": 10,
```
Press `[CTRL]+[X]` at the same time, then `[Y]` to confirm. Lastly, reboot with `sudo reboot` and it should show the border. 

To help identifying what caused this issue, may I ask you to upload your settings.json file with any sensitive data removed? Then I can test the very same file locally to figure out why it happened in the first place 👍

jamesvcc, 2023-11-10:
Thank you – your suggestion worked.

The two values were set at:

"padding_x": 10,

"padding_y": 20,

I changed them as suggested and it now works.  Do you still want to see my settings.json file?

Best Regards,
James

aceisace, 2023-11-10:
Glad to hear it worked 😄 
In this case, the issue is clearly with the paddings so no need to post the settings.json file  👍 
Closing this issue as the problem has been resolved

closed 2023-11-10 by aceisace