aceinnolab/Inkycal #281: Red circle and fram off center since 2.0.3

mygrexit, 2023-11-10:
**Describe the bug**
The red circle and the border lines are off center. Image attached.

I can only speak for the 12.48 waveshare display.

**To Reproduce**
Steps to reproduce the behavior:

Install new update.

**Expected behavior**

As before. Centered frame and centered red dot.

**Screenshots**

Photo.

[image]

aceisace, 2023-11-24:
@mygrexit May I ask you to test the issue with the latest version of Inkycal? 
To update, you can run:
```sh
sudo apt-get install wkhtmltopdf -y
cd ~/Inkycal
git pull
source venv/bin/activate
pip install -r requirements.txt
# To test if Inkycal is still running fine:
python inky_run.py
```

If all works well, you can then reboot and check if the bug still remains

mygrexit, 2023-11-24:
Unfortunately not @aceisace, it looks bettet but it is still off. Going to send a picture tommorow since i don't have a phone near me.

mygrexit, 2023-11-25:
Here you go @aceisace 
[image]
[image]

mygrexit, 2023-11-25:
Here is how it looks a single module

[image]

aceisace, 2023-11-25:
Thanks for the feedback @mygrexit . Thanks to these images, I can figure out roughly where to look for the issue 👍. I'll push a fix for this in a separate branch before merging into main

mygrexit, 2024-02-08:
Hello @aceisace, have you had the chance to have a look or is it a bug you won't fix due to the rewrite? :)

aceisace, 2024-02-08:
Apologies for the late reply. I did have a chance to look into this just now and will push the fix right away. Once the fix is pushed, please run `git pull` in the Inkycal folder and run `sudo reboot` to restart Inkycal. Please let me know it this fixes the issue as I can confirm it works on my end

closed 2024-02-08 by aceisace

SIDy1993, 2024-02-08:
[image]

mygrexit, 2024-03-07:
I can confirm that it looks like in @SIDy1993 comment. Maybe reopen the bug @aceisace :-D