aceinnolab/Inkycal #283: Smushed / cut-off text

andres00arroyo, 2023-11-20:
**Describe the bug**
Text on the screen is being cut-off vertically.

**To Reproduce**
Steps to reproduce the behavior:
1. raspberry pi zero 2W + raspbian OS [Legacy] + 7.5" Waveshare B&W (VER 2)
2. Install Inkycal
3. Configure JSON per the settings at the bottom of this submission
4. Start inkycal
5. observe smushed calendar text (also todoist text is smushed)

**Expected behavior**
Not smushed (example: g, p, q, y, and any letters with an underhang do not properly show)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Raspberry Pi OS [LEGACY]

JSON FILE:
{
    "model": "epd_7_in_5_v2",
    "update_interval": 30,
    "orientation": 0,
    "info_section": false,
    "info_section_height": null,
    "border_around_modules": false,
    "calibration_hours": [
        0,
        12,
        18
    ],
    "modules": [
        {
            "position": 1,
            "name": "Agenda",
            "config": {
                "size": [
                    480,
                    240
                ],
                "ical_urls": "DELETED",
                "ical_files": null,
                "date_format": "dddMMMD",
                "time_format": "h:mmA",
                "padding_x": 14,
                "padding_y": 14,
                "fontsize": 16,
                "language": "en"
            }
        },
        {
            "position": 2,
            "name": "Todoist",
            "config": {
                "size": [
                    480,
                    240
                ],
                "api_key": "DELETED",
                "project_filter": "Inbox",
                "padding_x": 14,
                "padding_y": 14,
                "fontsize": 16,
                "language": "en"
            }
        },
        {
            "position": 3,
            "name": "Feeds",
            "config": {
                "size": [
                    480,
                    320
                ],
                "feed_urls": "https://xkcd.com/atom.xml",
                "shuffle_feeds": true,
                "padding_x": 14,
                "padding_y": 14,
                "fontsize": 16,
                "language": "en"
            }
        }
    ]
}

aceisace, 2023-11-21:
I think I've identified the issue. The issue is within the write-function which writes text on a given image. Due to the Pillow upgrade and some changes in how the font-height is calculated, the few pixels above the actual letter are ignored, leading to shorter line-heights. Since the text does not fully fit in there anymore, the text appears to be cut-off. 

To fix this issue, I'm creating a branch and will provide instructions for you to test once I have pushed the fix on the branch 👍

andres00arroyo, 2023-11-21:
sounds good! Let me know when you need help testing 🙂

I'm finalizing the 3D frame I've made and will provide the file once it's perfected (there are still some minor issues with the current design)

aceisace, 2023-11-21:
Thanks for willing to help! Sure, looking forward to the 3d-printed frame once it's finalised. 

You can now test the potential fix:
```sh
cd ~/Inkycal
git pull
git checkout feature/#283
source venv/bin/activate
python3 inky_run.py
```
Please let me know if it works

aceisace, 2023-11-22:
Managed to test this myself and the line spacing has been restored to how it should look like. Creating PR for merge 👍

andres00arroyo, 2023-11-22:
Oh good stuff! Traveling for the holiday’s at the moment - I finished up the design but will need to wait until Saturday to send over the file.Andres Arroyo484 797 4732

aceisace, 2023-11-22:
Sure, enjoy your holidays :D . No need to rush with the design

closed 2023-11-23 by aceisace