aceinnolab/Inkycal #288: [FEATURE]: Flexible Calibration Hours Setting in WEB-UI

mygrexit, 2023-11-24:
**Overview**

The current implementation of Inkycals settings.json through the WEB-UI requires users to set three specific times for display calibration (calibration_hours). This feature request proposes an enhancement to make the calibration hours setting more flexible in the WEB-UI.

**Proposed Change**
Allow users to input a variable number of calibration hours, including the option to set none. This change will enable users to have more control over calibration scheduling, including the ability to disable automatic calibration completely if no time is specified.

**UI Update:**

Modify the calibration hours input section in the WEB-UI to allow for:
The option to leave the calibration hours empty, which should be interpreted as disabling automatic calibration.

**Backend Adjustment:** 

Ensure that the backend logic that generates settings.json can handle:

- An** array of any number of integers for calibration hours.
- An empty array, which should disable automatic calibration.

aceisace, 2023-12-13:
I've implemented this feature in the web-ui, may I ask you to test if setting the calibration hours to "skip" does indeed skip calibration and that there are no errors before I close this issue?

mygrexit, 2023-12-14:
Hey @aceisace,

thanks for implementing that!

I tested it and couldn't find any problems. It worked for me.

aceisace, 2023-12-15:
Perfect, I've merged it into main now and will be closing this issue too 👍

closed 2023-12-15 by aceisace