aceinnolab/Inkycal #289: [FEATURE]: 16 grayscales support

Zer0great, 2023-12-06:
Hello, as Inkycal already supports the parallel Waveshare displays, it would be great if their 16 grayscale capability could be used too. Especially with the IMAGE or SLIDESHOW modules, as the display is capable of high quality grayscale output.

Currently there are only 2/3 color modes supported. As the parallel displays are quite high-res/dpi, photos do not look entirely bad even in just b/w dithered. Even so, fully utilizing the display capability in Inkycal would be a step forward.

Thank you.

aceisace, 2023-12-14:
@Zer0great Thanks for opening this issue. I've implemented a "gray16" palette here: #294 . Once the test run through, I will merge it into main and take care of the web-ui to support this new palette. 
As soon as this is completed, you can test it out with a `git pull` from the Inkycal folder, changing the palette in your `settings.json` to "gray16"

aceisace, 2023-12-15:
@Zer0great Could you confirm the implementation works as expected now so this issue can be closed?
Here are the instructions:
```
cd ~/Inkycal
git pull
# edit settings.json file. Your file may be located elsewhere. Change the path accordingly
sudo nano /boot/settings.json 
# set palette to `16gray`
# save the file with [CTRL]+[X], then press [Y] to confirm
# run Inkycal:
source venv/bin/activate
python inky_run.py
```

Zer0great, 2023-12-15:
Hello, @aceisace unfortunately I can not get it to work. Error is "TypeError: Display.render() missing 1 required positional argument: 'im_colour'"
As part of troubleshooting I tried complete manual installation from scratch, into different folder, creating new venv etc.
The same error happens with palette bw or 16gray - no difference.

Problem is with both modules - image, slideshow.

Please see attached settings.json
[settings.json](https://github.com/aceinnolab/Inkycal/files/13684831/settings.json)

(venv) berry@berry:~/Inkycal $ python3 inky_run.py
inkycal.modules.inkycal_image loaded
loaded inkycal
Inkycal version: v2.0.3

Selected E-paper display: 10_in_3
generating image(s) for Inkycal Image - show an image from a URL or local path...OK!
Inkycal version: v2.0.3

Selected E-paper display: 10_in_3
Date: 15 Dec 23 | Time: 12:50
Generating images for all modules...successful
Traceback (most recent call last):
  File "/home/berry/Inkycal/inky_run.py", line 7, in <module>
    asyncio.run(inky.run())  # If there were no issues, you can run Inkycal nonstop
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/berry/Inkycal/inkycal/main.py", line 368, in run
    display.render(im_black)
TypeError: Display.render() missing 1 required positional argument: 'im_colour'

aceisace, 2023-12-18:
Thanks for testing this, @Zer0great . I'll write a few unittests and will push a fix for the problem soon 👍

aceisace, 2023-12-19:
@Zer0great I found a fix for this error:
`TypeError: Display.render() missing 1 required positional argument: 'im_colour'`

Please run the following to fix the issue with this error:
```sh
cd ~/Inkycal
git pull
source venv/bin/activate
python inky_run.py
```

Zer0great, 2023-12-20:
Hello, I have tested the new version, now it basically works, but the output seems to be just black and white without gradation. Please see attached pictures - first is comparison of palette "BW" vs. "16gray", second is just bigger picture with "16gray". Sorry for bad quality, hopefuly it is adequate for demonstration.

[image]
[image]

closed 2025-09-03 by github-actions[bot]