aceinnolab/Inkycal #290: error

lukasMega, 2023-12-09:
Hello,
could you please check (and fix if possible) this line maybe:
https://github.com/aceinnolab/Inkycal/blob/83b20af2620d860134366481feaf3b700a2549d6/inkycal/main.py#L366C25-L366C49 ?

Seems it's missing second parameter `im_colour` because I got an error:

output:
```console
(venv) dietpi@RPi0paper:~/Inky-new/Inkycal$ python3 inky_run.py
inkycal.modules.inkycal_weather loaded
inkycal.modules.inkycal_calendar loaded
inkycal.modules.inkycal_feeds loaded
loaded inkycal
Inkycal version: v2.0.3

Selected E-paper display: epd_7_in_5_v2
generating image(s) for Weather (openweathermap) - Get weather forecasts from openweathermap...OK!
generating image(s) for Calendar - Show monthly calendar with events from iCalendars...OK!
generating image(s) for RSS / Atom - Display feeds from given RSS/ATOM feeds...OK!
Inkycal version: v2.0.3

Selected E-paper display: epd_7_in_5_v2
Date: 9 Dec 23 | Time: 22:17
Generating images for all modules...successful
Traceback (most recent call last):
```
```python
  File "/home/dietpi/Inky-new/Inkycal/inky_run.py", line 7, in <module>
    asyncio.run(inky.run())  # If there were no issues, you can run Inkycal nonstop
    ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/dietpi/Inky-new/Inkycal/inkycal/main.py", line 366, in run
    display.render(im_black)
TypeError: Display.render() missing 1 required positional argument: 'im_colour'
```

Thank you!

lukasMega, 2023-12-09:
update: I fixed it locally by adding `None` in second parameter:

```diff
- display.render(im_black)
+ display.render(im_black, None)
```

aceisace, 2023-12-09:
Thanks for reporting the bug and finding a way to fix it @lukasMega ! 
I'm creating a branch for the proper bugfix and will merge it shortly. This issue will be closed automatically along with the merge

lukasMega, 2023-12-09:
Thank you for very 🚀 fast response! 💪

aceisace, 2023-12-09:
You're welcome. Thank you too for finding this bug! The PR is created already and will take about 20 mins until all tests have run. After that it will be merged and should contain the bugfix. I'll ping you here asking you to test the bugfix to see if it really works as expected 👍

lukasMega, 2023-12-09:
By the way; In July, I upgraded Inkycal on my Raspberry Pi Zero W. However, the upgrade caused Inkycal to stop working. Due to the Raspberry Pi Zero W’s limited processing power (1 CPU, 1 thread), I was unable to troubleshoot the issue at that time. Recently, I gave it another try and the setup process took 1-2 hours. Fortunately, Inkycal is now functioning properly.

I applied your changes locally and I confirm it's working.
Thank you, and I hope you have a pleasant evening/night. 😄

closed 2023-12-09 by aceisace

aceisace, 2023-12-09:
Sorry to hear you had some difficulties with Inkycal in July! Inkycal is finally becoming more stable thanks to the dedicated workflow to ensure it works exactly as expected on Raspberry Pi OS. But implementing and testing such a workflow required a lot of time and research.
It's been a few months until the workflow has been implemented. Ever since, the number of problems related to the installation of Inkycal has also decreased drastically. You can therefore expect more stable releases and updates without having to worry of something breaking apart.

The fairly long installation time of Inkycal is related to the growing number of modules, which depend on python libraries. Therefore, there isn't a real way to shorten the installation time on the Zero W, but contributors and sponsors of Inkycal get access to InkycalOS-Lite, which has everything pre-installed and configured. 

I saw from the log that you're using DietPi. Is there a way to test this myself too? I want to test if DietPi is better suited to install Inkycal and see if there are speed improvements when running Inkycal. 

Last but not least, please pull the latest changes to check if the fix works. You can either reboot or run inky_run.py