aceinnolab/Inkycal #293: Inkcal crashing ...

jamesvcc, 2023-12-11:
Hi,

As of this morning, I executed a "git pull" and restarted my frame.  It has now stopped working with the following log errors:

08-12-2023 10:31:14 | inkycal.modules.inkycal_weather |  INFO: Image size: (460, 110)
08-12-2023 10:31:15 | inkycal.modules.inkycal_weather |  INFO: Connection test passed
Traceback (most recent call last):
  File "/home/inky/Inkycal/inky_run.py", line 2, in <module>
    from inkycal import Inkycal
  File "/home/inky/Inkycal/inkycal/__init__.py", line 5, in <module>
    import inkycal.modules.inkycal_agenda
  File "/home/inky/Inkycal/inkycal/modules/__init__.py", line 11, in <module>
    from .inkycal_webshot import Webshot
  File "/home/inky/Inkycal/inkycal/modules/inkycal_webshot.py", line 6, in <module>
    from htmlwebshot import WebShot
ModuleNotFoundError: No module named 'htmlwebshot'

jamesvcc, 2023-12-11:
Thanks for that, I have tried it and there were quite a few dependencies that were also installed.

Unfortunately, I still get the same error message.

aceisace, 2023-12-11:
Each commit of Inkycal runs through a dedicated pipeline which tests that the installation according to the README is still working on the set version of Raspberry Pi OS. You can try to re-install Inkycal like this. Before you do, please ensure your settings.json file is backed up somewhere:
```sh
# increase swap size:
sudo dphys-swapfile swapoff
sudo sed -i -E '/^CONF_SWAPSIZE=/s/=.*/=512/' /etc/dphys-swapfile
sudo dphys-swapfile setup
sudo dphys-swapfile swapon

# delete the current Inkycal folder:
cd $HOME && rm -r Inkycal

sudo apt-get install zlib1g libjpeg-dev libatlas-base-dev rustc libopenjp2-7 python3-dev scons libssl-dev python3-venv python3-pip git libfreetype6-dev wkhtmltopdf
cd $HOME
git clone --branch main --single-branch https://github.com/aceinnolab/Inkycal
cd Inkycal
python3 -m venv venv
source venv/bin/activate
python -m pip install --upgrade pip
pip install wheel RPi.GPIO==0.7.1 spidev==3.5
pip install -e ./

# then you can test if Inkycal is working fine:
cd $HOME/Inkycal
source venv/bin/activate
python3 inky_run.py
```

aceisace, 2023-12-17:
Have you managed to get it running yet @jamesvcc ?

jamesvcc, 2023-12-17:
Sorry, no. I have been away and will be away for the next week, but I know it is still not working with the same error. It is an InkyOS install. I will look at it again after Xmas.

Best Regards
James

aceisace, 2023-12-19:
I double checked this issue and it seems this should fix it:
```sh
sudo apt-get install wkhtmltopdf -y
cd ~/Inkycal
source venv/bin/activate
pip install htmlwebshot
```

Please let me know if this works for you too

jamesvcc, 2023-12-28:
Hi, 

Sorry for the delay in getting back to you.  Yes, I did try your suggestion above.  It installed quite a few items and then left some other dependencies I had to install - but it now works.

Thanks for the assistance.

closed 2023-12-28 by jamesvcc