aceinnolab/Inkycal #295: [FEATURE]: Flip image in settings.json

Zer0great, 2023-12-14:
On my Waveshare 10in3 display, Inkycal is left-right flipped (mirror image).
Please implement simple image flip setting in settings.json , that would trigger for instance "image.transpose(method=Image.FLIP_LEFT_RIGHT)" somewhere in the image generation process.
Thank you

aceisace, 2023-12-14:
@Zer0great Thanks for submitting this issue! 
As this is specific for the 10.3" display, the "official" approach would be to hardcode this into the driver file of the 10.3" display. While the web-UI does support flipping the image upside down, mirroring will take longer to be implemented (more changes across several files). 
I'll push a fix for this soon and would like to hear your feedback after testing 👍

aceisace, 2023-12-14:
Fix implemented, but not tested yet. I will merge the PR and will ask you to test it to ensure it actually works as expected

closed 2023-12-15 by aceisace