aceinnolab/Inkycal #299: Temporary failure in name resolution while obtaining image from web address

Atrejoe, 2023-12-19:
**Describe the bug**
A possibly transient error (failure to resolve the address), possibly unhandled.

**To Reproduce**
Use an image panel, specify a url of a non-exitent domain (in my case the domain name existed).

**Expected behavior**
Error message to appear on the panel (possibly as overlay), retry to obtain the image after pause, not crash the process (unsure if this happens)

**Screenshots**
n/a

**Desktop (please complete the following information):**
Ace-provide OS image

**Smartphone (please complete the following information):**
n/a

**Additional context**
Calender has been running smoothly for a few days.

``` txt
19-12-2023 16:41:00 | inkycal.modules.inkycal_image |  INFO: Image size: (528, 880)
19-12-2023 16:41:00 | inkycal.modules.inky_image |  INFO: inkycal.modules.inky_image loaded
19-12-2023 16:41:00 | inkycal.modules.inky_image |  INFO: loading image from URL
19-12-2023 16:41:30 | inkycal.modules.inky_image |  ERROR: Invalid Image file provided
Traceback (most recent call last):
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connection.py", line 203, in _new_conn
    sock = connection.create_connection(
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/util/connection.py", line 60, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 790, in urlopen
    response = self._make_request(
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 491, in _make_request
    raise new_e
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 467, in _make_request
    self._validate_conn(conn)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 1096, in _validate_conn
    conn.connect()
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connection.py", line 611, in connect
    self.sock = sock = self._new_conn()
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connection.py", line 210, in _new_conn
    raise NameResolutionError(self.host, self, e) from e
urllib3.exceptions.NameResolutionError: <urllib3.connection.HTTPSConnection object at 0xaf22bd00>: Failed to resolve 'inkycal.robertsirre.nl' ([Errno -3] Temporary failure in name resolution)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/adapters.py", line 486, in send
    resp = conn.urlopen(
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/connectionpool.py", line 844, in urlopen
    retries = retries.increment(
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/urllib3/util/retry.py", line 515, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='inkycal.robertsirre.nl', port=443): Max retries exceeded with url: /panel/[MY SECRET PANEL ID] (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0xaf22bd00>: Failed to resolve 'inkycal.robertsirre.nl' ([Errno -3] Temporary failure in name resolution)"))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/inky/Inkycal/inkycal/modules/inky_image.py", line 52, in load
    image = Image.open(requests.get(path, stream=True).raw)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/home/inky/Inkycal/venv/lib/python3.9/site-packages/requests/adapters.py", line 519, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='inkycal.robertsirre.nl', port=443): Max retries exceeded with url: /panel/[MY SECRET PANEL ID] (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0xaf22bd00>: Failed to resolve 'inkycal.robertsirre.nl' ([Errno -3] Temporary failure in name resolution)"))

```

aceisace, 2023-12-19:
Hi there @Atrejoe , 
thanks for reporting this issue! I'm assuming the error is caused by the a URL not starting with http/https. I can update the image module of Inkycal to behave in this way:
1) Check if the location of the image is local, i.e. starts with `/`. If yes, check if it exists. If yes, continue.
2) If case 1 is not true, check if the URL starts with http/https. If not, raise an AssertionError: (Please check if the URL starts with http/https)
3) Check if the URL can actually be obtained from the valid URL. If not, raise AssertionError (Could not fetch an image from the given URL)

Does this sound like a solution?

Atrejoe, 2023-12-19:
Hey @aceisace , excellent response time!

No, this is not the case: the url was actually a correct url, and did not change over time (it has been running fine untouched for days).
I this this is actually a network issue, either it looses internet connection or the DNS server was somehow unreachable. An incorrect DNS record (for instance if the underlying ip-address had been changed) would likely have resulted in another kind of error.

I think your proposed detection mechanism for input validation was once partially present in the pull request I made and is good to implement anyway. (when starting with http(s), assume it's a url and [validate the format](https://stackoverflow.com/a/32171869/201019), otherwise validate if the path is valid and the target exists)

If you want a test-url, you could use this:
https://inkycal.robertsirre.nl/panel/test/epd_7_in_5_v3_colour/panel-of-panels

aceisace, 2023-12-19:
I've checked the url from my mac and it seems to work just fine, however, I do remember that this issue could be related to DNS/Firewall/WiFi-connectivity too. It's hard to pinpoint what exactly was causing this issue, but maybe we could add a retry after a few seconds to see if it resolves the error? 
For example, if a URL was detected, try to download the image, if that fails, retry again after n seconds. Repeat this up to three times before giving up. Does this approach sound more reasonable? 

I just check the thing with validation too, it's already implemented at the download/load stage, but I've added it to the init function of this module too. Will be pushed with the next commit.

closed 2024-03-19 by github-actions[bot]